istoreos/istoreos #1507: 用iStore OS拨号最近经常断线

Jernin, 2024-04-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
光猫桥接拨号现在经常性断网。
### (1) 具体问题
光猫桥接拨号现在经常性断网，短则半小时长的话1小时左右。而且断线会出先拨号不成功的情况。重新安装最新系统也是不行
A：

### (2) 路由器型号和固件版本
X86软路由  ESXI8.0   固件版本
A：

### (3) 详细日志和/或截图
附重做系统以后的完整日志。
Thu Apr 18 00:03:04 2024 daemon.info pppd[18634]: No response to 5 echo-requests
Thu Apr 18 00:03:04 2024 daemon.notice pppd[18634]: Serial link appears to be disconnected.
Thu Apr 18 00:03:04 2024 daemon.info pppd[18634]: Connect time 23.3 minutes.
Thu Apr 18 00:03:04 2024 daemon.info pppd[18634]: Sent 5464945 bytes, received 41830418 bytes.
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Withdrawing address record for 106.118.2.46 on pppoe-wan.
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 106.118.2.46.
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Thu Apr 18 00:03:04 2024 daemon.notice ttyd[14559]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 00:03:04 2024 daemon.notice ttyd[14559]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Thu Apr 18 00:03:04 2024 daemon.err odhcp6c[18884]: Failed to send RS (Network unreachable)
Thu Apr 18 00:03:04 2024 daemon.err odhcp6c[18884]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Interface pppoe-wan.IPv6 no longer relevant for mDNS.
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address 240e:340:9907:523d:2daf:9dff:5b95:1fff.
Thu Apr 18 00:03:04 2024 daemon.notice ttyd[14559]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Withdrawing address record for 240e:340:9907:523d:2daf:9dff:5b95:1fff on pppoe-wan.
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Interface 'wan' has lost the connection
Thu Apr 18 00:03:04 2024 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Thu Apr 18 00:03:04 2024 daemon.notice netifd: wan_6 (18884): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Thu Apr 18 00:03:04 2024 daemon.err odhcp6c[18884]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Interface 'wan_6' is now down
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Registering new address record for fe80::20c:29ff:fef4:1248 on br-lan.*.
Thu Apr 18 00:03:04 2024 daemon.info avahi-daemon[10702]: Registering new address record for fd09:7cda:46a::1 on br-lan.*.
Thu Apr 18 00:03:04 2024 daemon.notice netifd: Interface 'wan_6' is disabled
Thu Apr 18 00:03:05 2024 daemon.err nmbd[19786]: [2024/04/18 00:03:05.128868,  0] ../../source3/nmbd/nmbd.c:59(terminate)
Thu Apr 18 00:03:05 2024 daemon.err nmbd[19786]:   Got SIGTERM: going down...
Thu Apr 18 00:03:05 2024 daemon.err nmbd[8194]: [2024/04/18 00:03:05.151443,  0] ../../source3/nmbd/nmbd.c:901(main)
Thu Apr 18 00:03:05 2024 daemon.err nmbd[8194]:   nmbd version 4.18.8 started.
Thu Apr 18 00:03:05 2024 daemon.err nmbd[8194]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Thu Apr 18 00:03:05 2024 daemon.err smbd[8193]: [2024/04/18 00:03:05.162949,  0] ../../source3/smbd/server.c:1746(main)
Thu Apr 18 00:03:05 2024 daemon.err smbd[8193]:   smbd version 4.18.8 started.
Thu Apr 18 00:03:05 2024 daemon.err smbd[8193]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Thu Apr 18 00:03:06 2024 daemon.warn dnsmasq[1]: ignoring query from non-local network fd72:3f6d:4900:7346:d15a:cbdb:9a72:af49 (logged only once)
Thu Apr 18 00:03:10 2024 daemon.notice pppd[18634]: Connection terminated.
Thu Apr 18 00:03:10 2024 daemon.info pppd[18634]: Connect time 23.3 minutes.
Thu Apr 18 00:03:10 2024 daemon.info pppd[18634]: Sent 5464945 bytes, received 41830418 bytes.
Thu Apr 18 00:03:10 2024 daemon.debug ModemManager[8350]: hotplug: event reported: action=remove, name=pppoe-wan, subsystem=net
Thu Apr 18 00:03:10 2024 daemon.notice pppd[18634]: Modem hangup
Thu Apr 18 00:03:10 2024 daemon.info pppd[18634]: Exit.
Thu Apr 18 00:03:10 2024 daemon.notice netifd: Interface 'wan' is now down
Thu Apr 18 00:03:10 2024 daemon.notice netifd: Interface 'wan' is setting up now
Thu Apr 18 00:03:10 2024 daemon.err insmod: module is already loaded - slhc
Thu Apr 18 00:03:10 2024 daemon.err insmod: module is already loaded - ppp_generic
Thu Apr 18 00:03:10 2024 daemon.err insmod: module is already loaded - pppox
Thu Apr 18 00:03:10 2024 daemon.err insmod: module is already loaded - pppoe
Thu Apr 18 00:03:10 2024 daemon.info pppd[8424]: Plugin pppoe.so loaded.
Thu Apr 18 00:03:10 2024 daemon.info pppd[8424]: PPPoE plugin from pppd 2.4.9
[... 252 lines cut ...]
Thu Apr 18 00:07:08 2024 daemon.info avahi-daemon[10702]: Registering new address record for 240e:342:9954:cc00::1 on br-lan.*.
Thu Apr 18 00:07:08 2024 daemon.info avahi-daemon[10702]: Withdrawing address record for 240e:342:9954:a900::1 on br-lan.
Thu Apr 18 00:07:08 2024 daemon.info avahi-daemon[10702]: Withdrawing address record for fe80::20c:29ff:fef4:1248 on br-lan.
Thu Apr 18 00:07:08 2024 daemon.info avahi-daemon[10702]: Withdrawing address record for fd09:7cda:46a::1 on br-lan.
Thu Apr 18 00:07:08 2024 daemon.notice netifd: Interface 'wan_6' is now up
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using nameserver 222.222.222.222#53
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using nameserver 222.222.202.202#53
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using nameserver 240e:4c:4008::1#53
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using nameserver 240e:4c:4808::1#53
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Apr 18 00:07:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Thu Apr 18 00:07:08 2024 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]: [2024/04/18 00:07:30.149463,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]:   *****
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]:
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]:   Samba name server ISTOREOS is now a local master browser for workgroup WORKGROUP on subnet 10.0.0.1
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]:
Thu Apr 18 00:07:30 2024 daemon.err nmbd[16212]:   *****
Thu Apr 18 00:17:10 2024 user.warn ddns-scripts[15574]: Jernin: Updating IP at DDNS provider failed - starting retry 1/0

bailuxianfeiying, 2024-04-20:
有可能是光猫问题，我家的光猫桥接模式也频繁断线

jjm2473, 2024-05-10:
之前有用户反馈也是光猫问题，找运营商换光猫就好了

closed 2024-05-10 by jjm2473