istoreos/istoreos #1511: docker容器内部无法访问外网IPV6

Mick-Mi, 2024-04-21:
最近在istore os docker中搭建了wg-easy. 方便手机快速回家组网. 不过发现docker容器内无法访问ipv6地址. 然后在网上找了很多参考. 通过设置 `/etc/config/dockerd` 
```
root@Openwrt:~# cat /etc/config/dockerd
# The following settings require a restart of docker to take full effect, A reload will only have partial or no effect:
# log_driver
# bip
# blocked_interfaces
# extra_iptables_args
# device

config globals 'globals'
	option alt_config_file '/etc/docker/daemon.json'
```
以及`/etc/docker/daemon.json` 

```
{
    "ipv6": true,
    "fixed-cidr-v6": "fd00::/80",
    "experimental": true,
    "ip6tables": true
}
```
成功在docker中开启了ipv6 和 ipv6 nat 

```
[
    {
        "Name": "bridge",
        "Id": "be25dd9555d6fd08c610348d40d028750d9c7f81c777da77603f83a0b9b95992",
        "Created": "2024-04-18T18:03:51.298691989Z",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": true,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.17.0.0/16",
                    "Gateway": "172.17.0.1"
                },
                {
                    "Subnet": "fd00::/80",
                    "Gateway": "fd00::1"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "970f70383fb00d007ba7a99b9535d1508c64db8b1e21aebc95786738727e5780": {
                "Name": "wireguard",
                "EndpointID": "d9f5aed1909ec295be4f249671b3003ea10b95349d95a00638affab61e47992f",
                "MacAddress": "02:42:ac:11:00:02",
                "IPv4Address": "172.17.0.2/16",
                "IPv6Address": "fd00::242:ac11:2/80"
            }
        },
        "Options": {
            "com.docker.network.bridge.default_bridge": "true",
            "com.docker.network.bridge.enable_icc": "true",
            "com.docker.network.bridge.enable_ip_masquerade": "true",
            "com.docker.network.bridge.host_binding_ipv4": "0.0.0.0",
            "com.docker.network.bridge.name": "docker0",
            "com.docker.network.driver.mtu": "1500"
        },
        "Labels": {}
    }
]

```

```

root@iStoreOS:~# ip6tables -t nat -nL
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
ACCEPT     tcp      ::/0                 2001:4860:4860::8844  /* OpenClash Google DNS Hijack */ tcp dpt:53
ACCEPT     tcp      ::/0                 2001:4860:4860::8888  /* OpenClash Google DNS Hijack */ tcp dpt:53
REDIRECT   tcp      ::/0                 ::/0                 tcp dpt:53 /* OpenClash DNS Hijack */ redir ports 53
REDIRECT   udp      ::/0                 ::/0                 udp dpt:53 /* OpenClash DNS Hijack */ redir ports 53
DOCKER     all      ::/0                 ::/0                 ADDRTYPE match dst-type LOCAL
REDIRECT   udp      ::/0                 ::/0                 /* Dnsmasq */ udp dpt:53 redir ports 53

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
DOCKER     all      ::/0                !::1                  ADDRTYPE match dst-type LOCAL

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all      fd00::/80            ::/0
MASQUERADE  tcp      fd00::242:ac11:2     fd00::242:ac11:2     tcp dpt:51821
MASQUERADE  udp      fd00::242:ac11:2     fd00::242:ac11:2     udp dpt:51820

Chain DOCKER (2 references)
target     prot opt source               destination
RETURN     all      ::/0                 ::/0
DNAT       tcp      ::/0                 ::/0                 tcp dpt:51821 to:[fd00::242:ac11:2]:51821
DNAT       udp      ::/0                 ::/0                 udp dpt:51820 to:[fd00::242:ac11:2]:51820
```

现在的情况是 docker 容器可以ping 通宿主机的ipv6, 宿主机本身也能ping通外网的ipv6, 但是docker无法ping外网ipv6

bridge network ping 外网
```
root@iStoreOS:~# docker run --network=bridge --rm -it busybox ping -6 -c4 taobao.com
PING taobao.com (2408:4001:f00::27c): 56 data bytes

--- taobao.com ping statistics ---
4 packets transmitted, 0 packets received, 100% packet loss
```

host 模式ping 外网 
```
root@iStoreOS:~# docker run --network=bridge --rm -it busybox ping -6 -c4 taobao.com
PING taobao.com (2408:4001:f00::27c): 56 data bytes

--- taobao.com ping statistics ---
4 packets transmitted, 0 packets received, 100% packet loss
root@iStoreOS:~# docker run --network=host --rm -it busybox ping -6 -c4 taobao.com
PING taobao.com (2408:4002:1f30::a5): 56 data bytes
64 bytes from 2408:4002:1f30::a5: seq=0 ttl=89 time=40.113 ms
64 bytes from 2408:4002:1f30::a5: seq=1 ttl=89 time=38.933 ms
64 bytes from 2408:4002:1f30::a5: seq=2 ttl=89 time=40.163 ms
64 bytes from 2408:4002:1f30::a5: seq=3 ttl=89 time=40.358 ms

--- taobao.com ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 38.933/39.891/40.358 ms
```

bridge 模式ping 宿主机   宿主机IP 稍微处理过了 
```
root@iStoreOS:~# docker run --network=bridge --rm -it busybox ping -6 -c4 [240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828]
PING [240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828] (240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828): 56 data bytes
64 bytes from 240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828: seq=0 ttl=64 time=0.165 ms
64 bytes from 240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828: seq=1 ttl=64 time=0.217 ms
64 bytes from 240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828: seq=2 ttl=64 time=0.219 ms
64 bytes from 240e:xxx:61:xxxx:xxxx:e1ff:fe13:b828: seq=3 ttl=64 time=0.223 ms

--- [240e:39c:61:cfb1:7e2b:e1ff:fe13:b828] ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 0.165/0.206/0.223 ms
```

相同的操作  我在Ubuntu虚拟机中能够实现 在docker容器内部中访问 ipv6 外网.  但是istore os 始终无法访问, 有大佬能够答疑解惑吗

138161, 2024-04-21:
bridge network 是不是相当于和下面的pc一样了？大概是路由表的配置，要自己配吧，wan6口的ipv6一定没有PD前缀，只有ipv6地址

Mick-Mi, 2024-04-21:
> bridge network 是不是相当于和下面的pc一样了？大概是路由表的配置，要自己配吧，wan6口的ipv6一定没有PD前缀，只有ipv6地址

忘记补充了  我istore os  是作为旁路由用的

138161, 2024-04-21:
必须host模式，bridge network相当于光猫下面接了个路由器（istore），再接（bridge network）docker，icmpv6消息要穿越二层，消息能传递，应该能获得ipv6地址，但没法自己管理istore的路由表（因为bridge network）

138161, 2024-04-21:
或者 istore 增加这样的文件，将istore下面的docker，指向wan6侧
编辑文件  vim /etc/hotplug.d/iface/90-ipv6
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
GWaddr=`ifconfig | ip -6 route show default | head -n1 | awk '{print $5}'`
route -A inet6 add default gw $GWaddr dev eth0

Mick-Mi, 2024-04-21:
好奇 为啥 Ubuntu 那就简单配置下就可以直接访问了  两边处理 数据的逻辑不太一样么   感觉您的这个配置方法已经超过我能理解的范畴了

138161, 2024-04-22:
Ubuntu 大概替你维护了路由表，自己指向了吧

138161, 2024-04-22:
早上起来又看了下你的配置，docker的ipv6配置，实际上就是ipv6的NAT方式，
1.5、DHCP服务器--IPv6设置
     指定的主接口：不择
     RA 服务：服务器模式
     DHCPv6 服务：服务器模式
     NDP 代理：禁用
1.6、DHCP服务器--IPv6 RA设置
     默认路由器：强制的
    启用SLAAC：选择
    RA标记：受管配置 其他配置
第三步，网络--接口--全局网络选项
     IPv6 ULA 前缀：fd00::/64
     数据包引导：选择
第五步，网络--防火墙
--自定义规则

# ipv6 NAT
ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
ip6tables -A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT

ip6tables -I input_rule           -p udp   --sport 546 --dport 547  -j ACCEPT

ip6tables -I input_rule   -p ipv6-nonxt -m length --length 40 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 1  -m limit --limit 100/s --limit-burst 10  -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 2  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 3  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 4  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 128  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 129  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 130  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

ip6tables -I forwarding_wan_rule -p ipv6-nonxt -m length --length 40 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

Mick-Mi, 2024-04-23:
感谢大佬分享  我去试试

138161, 2024-04-23:
我看你会搞晕的
总之，第一步，要搞出fe80开头的ipv6，NAT模式
第二步，搞定路由表，能上外网
第三步，过几天不断网，搞定icmpv6消息传递，546 547端口
如果你能成功，代表大概对ipv6配置搞得差不多明白了

Mick-Mi, 2024-04-23:
确实晕了，我争取搞明白。 确实也不是这专业的，只是个人瞎研究。还是感谢大佬

closed 2024-04-28 by Mick-Mi