istoreos/istoreos #1512: 安装软件包时出现缺少“ucode-mod-lua”错误，请问该怎么解决？

draftbottle, 2024-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在istore中安装了几个插件后就会出现此问题，该问题会随机出现在某一个插件安装的过程中。在重新刷完固件后再次安装这个插件又会成功安装(不会报错)并正常使用。有时还会在卸载某插件后再次安装原插件报该错误的情况。而且opkg安装软件包时也会出现该情况。请问是不是某些插件有冲突啊？是安装太多插件导致ucode-mod-lua库丢失吗？

### (2) 路由器型号和固件版本

A：树莓派4B

### (3) 详细日志和/或截图

A：
[image]
[image]

closed 2024-04-22 by jjm2473

draftbottle, 2024-04-22:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

AlanWang2333, 2024-04-22:
> 估计你用了不知道什么版本的第三方软件仓库导致的版本错乱，ucode-mod-lua 是23.05才引入的。https://github.com/openwrt/luci/tree/openwrt-23.05/contrib/package/ucode-mod-lua

大佬牛逼，我也有这个问题，现在我把软件包的第三方仓库的src/gz openwrt_dllkids https://op.dllkids.xyz/packages/x86_64/  注释掉之后，就能安装软件了，为了这个我都不知道重置多少次了，哭死，我以为是装的软件多了自然就有冲突了。
大佬再帮忙看下这个仓库可以添加吗src/gz openwrt_kiddin9  https://dl.openwrt.ai/packages-23.05/x86_64/kiddin9/
[image]

jjm2473, 2024-04-23:
一般都不建议添加第三方的仓库，因为你不知道他用哪个版本sdk编译的。
如果实在要安装第三方仓库的软件，那么如果安装其他软件包出问题时注释掉再试，之后要安装第三方仓库的软件时再取消注释。

但是用了第三方仓库要是出问题就别反馈，自己解决。

如果你只是为了安装passwall或者其他翻墙插件，可以下载网友编译好的离线包 https://github.com/bcseputetto/Are-u-ok

AlanWang2333, 2024-04-23:
好的，谢谢😊一开始啥都不懂，看见人家出教程就往上加了

draftbottle, 2024-04-23:
好的，谢谢。😊

jjm2473, 2024-04-23:
有些第三方仓库只是给他们自己的固件用的，只能保证在他们的固件上没问题。

目前只有iStore提供的软件包会尽量兼容不同版本的固件，所以不使用第三方仓库的情况下，从iStore安装插件通常不会有问题。