istoreos/istoreos #1513: 海外用户无法安装iStore软件，和更新软件源，均提示SSL错误

bin1991, 2024-04-22:
[x] 没有类似的issue

## 详细叙述

### (1) 具体问题

A：无法在商店安装软件，提示SSL错误，经过相关操作后，更新系统中的openssl组件，我本地网络可以打开，也可以下载到 Packages.gz 这个文件，这个错误也导致软件源配置好后无法更新软件列表

### (2) 路由器型号和固件版本

A：X86_64，固件版本 22.03

### (3) 详细日志和/或截图

A：日志：
istore (失败于： 2024/4/22 20:01:07) > is-opkg install 'app-meta-webvirtcloud'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to istore.linkease.com:443 

截图：
[image]

[image]

jjm2473, 2024-04-22:
你用的什么固件？是官方istoreos吗？系统时间对不对？

jjm2473, 2024-04-22:
你自己的浏览器能否打开 https://istore.linkease.com/repo/

bin1991, 2024-04-23:
浏览器可以打开，但是，路由器本身打开就会错误，但是其他ssl会正常
[image]

bin1991, 2024-04-23:
> 你为什么不升级固件呢？怎么还在使用23年的固件，也许升级到新固件就好了，可能只是以为ssl版本太老或者CA证书没更新。

哦哦，好的，那升级一下就可以了对吧

jjm2473, 2024-04-23:
嗯，如果你只是想测试这个问题有没有解决，可以使用ventoy来测试新版本，比较灵活，随时可以换版本。