istoreos/istoreos #1514: 使用新版本istoreos 隔两天就断网 需要重启才可以上网

wybinhua, 2024-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用这个版本 上网大概两天到三天 就自动断网 需要重启设备后网络恢复

### (2) 路由器型号和固件版本

A：j4125 直接安装istoreos  最新版本

### (3) 详细日志和/或截图

A：

138161, 2024-04-22:
ip6tables -I input_rule -p udp --sport 546 --dport 547 -j ACCEPT

解决 DHCPv6 续约的 icmpv6 通讯，lan 侧到路由器本身，此消息再经过中继后，传到上级光猫。
过程：Solicit---Advertise---Request---Reply
PC 端口 546 ，发送请求 Solicit 至，路由器本身[ff02::1:2]:547 端口
路由器本身 546 端口，中继到，光猫 547 端口
光猫 547 端口，回应一个 Advertise 给，路由器 546 端口
路由器本身 547 端口，中继到，PC 端口 546
PC 端口 546 ，回应一个 Request 给路由器 547 端口
路由器本身 546 端口，中继到，光猫 547 端口
光猫 547 端口，回应一个 Reply 给，路由器 546 端口结束。

适用于 openwrt 和华硕路由器

再一个就是有 ipv6 地址，却无法上网
这是 openwrt 网关的问题，原因是未分配到 PD 前缀，只有一个/64 的地址，，此时就需要用户自己管理 openwrt 的路由表，具体方法
https://github.com/istoreos/istoreos/issues/1480

138161, 2024-04-22:
ip6tables -I input_rule   -p ipv6-nonxt -m length --length 40 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 1  -m limit --limit 100/s --limit-burst 10  -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 2  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 3  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 4  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 128  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 129  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 130  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I forwarding_wan_rule -p ipv6-nonxt -m length --length 40 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
#ip6tables -I forwarding_wan_rule -i eth0 -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

wldxx888, 2024-04-23:
我的也是，电信单线多拨，过两三天自动断网，然后就一直重拨不上，反复不断重拨，有时候要手动重启几次才能拨号成功

HamGuy, 2024-04-25:
我也遇到了，不过我的跟你的不太一样，我目前是隔段时间（具体未知，至少头天晚上正常，但是第二天白天就不行）就会出现 DNS 解析失败的情况，但是其他网络服务都正常, 猜测是DNS服务挂了

版本：iStoreOS 22.03.6 2024041213
设备：CncTion N4505-4L

nslookup 一直超时：
```shell
C:\Users\admin>nslookup baidu.com
DNS request timed out.
    timeout was 2 seconds.
Server:  UnKnown
Address:  192.168.0.110

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
*** Request to UnKnown timed-out
```

jjm2473, 2024-05-10:
@HamGuy 那试试换第三方的DNS，不要用运营商的

wybinhua, 2024-05-13:
我的更新最新版还是一样 打算换系统openwrt

QuantumLeap-us, 2024-05-31:
j4125同样的问题 更新了新版本 隔三岔五的DNS错误

jjm2473, 2024-06-04:
如果是dns服务挂了，可以在日志中看到，也就是dnsmasq进程。还可以用`ps w|grep dnsmasq|grep -v grep`命令检查进程在不在。如果真是dnsmasq挂了，大概率是某些翻墙/去广告插件导致的，关掉这些插件重启下，看看还会不会发生。

QuantumLeap-us, 2024-06-05:
我觉得是pw自带的 sing-box  xray 的过滤模式 和dnsmasq 冲突

jjm2473, 2024-06-05:
不需要猜测，看日志，一般这种配置问题导致挂掉，日志都会很清楚告诉你为什么。

HamGuy, 2024-06-05:
我刚才又遇到了，确实是 dnsmasq 服务挂掉了：
```shell

root@iStoreOS:~# ps w|grep dnsmasq -v
  PID USER       VSZ STAT COMMAND
    1 root      1424 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 IW<  [rcu_gp]
    4 root         0 IW<  [rcu_par_gp]
    6 root         0 IW<  [kworker/0:0H-ev]
    8 root         0 IW<  [mm_percpu_wq]
    9 root         0 SW   [rcu_tasks_trace]
   10 root         0 SW   [ksoftirqd/0]
   11 root         0 IW   [rcu_sched]
   12 root         0 SW   [migration/0]
   13 root         0 SW   [cpuhp/0]
   14 root         0 SW   [cpuhp/1]
   15 root         0 SW   [migration/1]
   16 root         0 SW   [ksoftirqd/1]
   18 root         0 IW<  [kworker/1:0H-ev]
   19 root         0 IW<  [netns]
  133 root         0 SW   [oom_reaper]
  138 root         0 IW<  [writeback]
  241 root         0 SW   [kcompactd0]
  357 root         0 IW<  [pencrypt_serial]
  359 root         0 IW<  [pdecrypt_serial]
  361 root         0 IW<  [cryptd]
  367 root         0 IW<  [kintegrityd]
  369 root         0 IW<  [kblockd]
  370 root         0 IW<  [blkcg_punt_bio]
  733 root         0 IW<  [ata_sff]
  734 root         0 SW   [watchdogd]
  862 root         0 IW<  [kworker/0:1H-kb]
  890 root         0 SW   [kswapd0]
  897 root         0 IW<  [kthrotld]
  945 root         0 IW<  [acpi_thermal_pm]
 1042 root         0 IW<  [iscsi_eh]
 1044 root         0 IW<  [iscsi_conn_clea]
 1049 root         0 IW<  [nvme-wq]
 1053 root         0 IW<  [nvme-reset-wq]
 1060 root         0 IW<  [nvme-delete-wq]
 1064 root         0 SW   [scsi_eh_0]
 1067 root         0 IW<  [scsi_tmf_0]
 1105 root         0 IW<  [uas]
 1157 root         0 IW<  [ipv6_addrconf]
 1179 root         0 IW<  [kworker/1:1H-kb]
 1187 root         0 IW<  [tpm_dev_wq]
 1277 root         0 SW   [jfsIO]
 1278 root         0 SW   [jfsCommit]
 1279 root         0 SW   [jfsCommit]
 1280 root         0 SW   [jfsSync]
 1283 root         0 IW<  [xfsalloc]
 1284 root         0 IW<  [xfs_mru_cache]
 1409 root         0 SW   [jbd2/nvme0n1p3-]
 1410 root         0 IW<  [ext4-rsv-conver]
 2224 ubus      1084 S    /sbin/ubusd
 2292 root       748 S    /sbin/askfirst /usr/libexec/login.sh
 2293 root       748 S    /sbin/askfirst /usr/libexec/login.sh
 2333 root       816 S    /sbin/urngd
 2641 root         0 IW<  [ena]
 2657 root         0 IW<  [md]
 2670 root         0 IW<  [raid5wq]
 2678 root         0 SW<  [aoe_tx0]
 2679 root         0 SW<  [aoe_ktio0]
 2701 root         0 IW<  [cifsiod]
 2707 root         0 IW<  [smb3decryptd]
 2708 root         0 IW<  [cifsfileinfoput]
 2709 root         0 IW<  [cifsoplockd]
 2721 root         0 IW<  [rpciod]
 2723 root         0 IW<  [kworker/u5:0-xp]
[... 156 lines cut ...]
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain dongtaiwang.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain mgoon.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain google.lu
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain bibox.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain brandonhutchinson.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain xinsheng.net
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain nflximg.net
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain zhinengluyou.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#15353 for domain guruonline.hk
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: more servers are defined but not logged
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bss.pandora.xiaomi.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using only locally-known addresses for o2o.api.xiaomi.com
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: using 13 more local addresses
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 13 names
Wed Jun  5 21:56:25 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

其中有一个异常 `Wed Jun  5 21:56:21 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM`, 我还以为是这个导致的，我这会儿重启后好了, 看了下日志，还是有这个异常，但是 dns 服务也是正常的，不是很理解啊

HamGuy, 2024-06-05:
> 不需要猜测，看日志，一般这种配置问题导致挂掉，日志都会很清楚告诉你为什么。

看了下系统日志，似乎不是很能懂，没发现是那里出的问题😓

jjm2473, 2024-06-06:
@HamGuy 你是不是搞错了，`ps w|grep dnsmasq -v`是排除掉了dnsmasq啊，不管有没有dnsmasq进程都不会显示，别误导人了。正确的命令是`ps w|grep dnsmasq|grep -v grep`。

HamGuy, 2024-06-06:
> @HamGuy 你是不是搞错了，`ps w|grep dnsmasq -v`是排除掉了dnsmasq啊，不管有没有dnsmasq进程都不会显示，别误导人了。正确的命令是`ps w|grep dnsmasq|grep -v grep`。

Sorry，命令搞错了，等下次出问题我再试试

xmcdlf1992, 2024-06-06:
> @HamGuy 你是不是搞错了，`ps w|grep dnsmasq -v`是排除掉了dnsmasq啊，不管有没有dnsmasq进程都不会显示，别误导人了。正确的命令是`ps w|grep dnsmasq|grep -v grep`。

大佬我的istore半个月断网两次了，配置是N100，只跑了lucky反向代理，clash等少数几个应用。网页和游戏全断开，网页提示找不到网站的DNS

jjm2473, 2024-06-06:
@xmcdlf1992 禁用或者卸载clash，然后重启下试试

xmcdlf1992, 2024-06-06:
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 nat table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Redirect '虚拟机-控制中心'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Redirect '小冷的游戏机'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * NAT 'DockerNAT'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: wan will enable FULLCONE-NAT
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 mangle table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 raw table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:      - Using automatic conntrack helper attachment
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 filter table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 nat table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 mangle table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 filter table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-DHCPv6'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-MLD'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Input'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Forward'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IPSec-ESP'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ISAKMP'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'linkease'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 nat table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_wan_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_wan_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_docker_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_docker_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_rule'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 mangle table
Thu Jun  6 19:30:23 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
[... cut ...]
Thu Jun  6 20:20:00 2024 cron.err crond[9830]: USER root pid 21917 cmd /etc/ShellCrash/task/task.sh 106 运行时每10分钟自动保存面板配置
Thu Jun  6 20:24:30 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.31.183 8a:a6:c0:5d:98:ec
Thu Jun  6 20:24:30 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.31.183 8a:a6:c0:5d:98:ec Xiaomi-13-Pro
Thu Jun  6 20:30:00 2024 cron.err crond[9830]: USER root pid 28243 cmd /etc/ShellCrash/task/task.sh 106 运行时每10分钟自动保存面板配置
Thu Jun  6 20:34:10 2024 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: nie.netease.com

xmcdlf1992, 2024-06-06:
> @xmcdlf1992 禁用或者卸载clash，然后重启下试试

日志我发在下面了，上一次断网时有openclash，断网之后我就换成了shellclash纯净模式（需要电脑手动设置代理ip+7890端口），然后开始又断网了，我在ssh里关闭clash重启就好了，能看看这是clash的原因吗？以前在AX9000和AX6上运行shellclash都不会断网呀
[log.txt](https://github.com/user-attachments/files/15620349/log.txt)

HamGuy, 2024-06-07:
> @HamGuy 你是不是搞错了，`ps w|grep dnsmasq -v`是排除掉了dnsmasq啊，不管有没有dnsmasq进程都不会显示，别误导人了。正确的命令是`ps w|grep dnsmasq|grep -v grep`。

刚才远程连回去，发现 DNS 又挂了，跑了下这个命令
``` shell
root@iStoreOS:~# ps w|grep dnsmasq|grep -v grep
21343 root      2160 S    {dnsmasq} /sbin/ujail -t 5 -n dnsmasq -u -l -r /bin/ubus -r /etc/TZ -r /etc/dnsmasq.conf -r /etc/ethers -
21350 dnsmasq  16660 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
```

不是很明白，看上去似乎 dnsmasq 服务没挂？
```
root@iStoreOS:~# ps w | grep dnsmasq
 2964 root      1124 S    grep dnsmasq
21343 root      2160 S    {dnsmasq} /sbin/ujail -t 5 -n dnsmasq -u -l -r /bin/ubus -r /etc/TZ -r /etc/dnsmasq.conf -r /etc/ethers -
21350 dnsmasq  16660 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
```

xmcdlf1992, 2024-06-15:
> @xmcdlf1992 禁用或者卸载clash，然后重启下试试

即使不卸载clash，重启后也能恢复，我现在卸载clash后跑了几天，暂时还没有出现断网（之前症状为打开网页全部提示DNS错误），难道istores不兼容clash吗

jjm2473, 2024-06-17:
openclash出问题很少见吗？

Nil-soo, 2025-08-07:
差不多，但是重启光猫就好了