istoreos/istoreos #1515: 关于单线多拨掉线和频繁重拨的问题

wldxx888, 2024-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我这里电信宽带，一直都是可以单线多拨（三拨）用了好多年了，之前一直使用LEDE固件2.33版本的单线多拨（这个版本的单线多拨非常好用，用了几年没有任何问题）。但目前istoreos版本的单线多拨却经常掉线，出现反复重拨，表现为虚拟的2个VWAN1和2口可以拨号成功，WAN口一直拨不上，或者偶尔WAN口拨上号也和前面两个VWAN端口间隔时间比较长，在1-3分钟左右。并且非常不稳定，过两三天后会掉线重拨，出现反复重拨。反复重拨表现为VWAN两个端口能拨上号，WAN口拨不上，导致VWAN两个端口拨上后也会掉线重拨，反复如此。

### (2) 路由器型号和固件版本

A：设备型号
INTEL Corporation Tiger Hill
固件版本
iStoreOS 22.03.6 2024031514
内核版本
5.10.201

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]

wldxx888, 2024-04-23:
(3) 详细日志和/或截图

Tue Apr 23 10:21:41 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/var/etc/shadowsocksr.include'
Tue Apr 23 10:21:42 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/var/etc/socat.include'
Tue Apr 23 10:21:42 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    ! Skipping due to path error: No such file or directory
Tue Apr 23 10:21:42 2024 daemon.err nmbd[18014]: [2024/04/23 10:21:42.856087,  0] ../../source3/nmbd/nmbd.c:901(main)
Tue Apr 23 10:21:42 2024 daemon.err nmbd[18014]:   nmbd version 4.18.8 started.
Tue Apr 23 10:21:42 2024 daemon.err nmbd[18014]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]: [2024/04/23 10:21:43.025648,  0] ../../source3/smbd/server.c:1746(main)
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]:   smbd version 4.18.8 started.
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]: [2024/04/23 10:21:43.074369,  0] ../../source3/smbd/server.c:1091(smbd_open_one_socket)
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]:   smbd_open_one_socket: open_socket_in failed: Address not available
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]: [2024/04/23 10:21:43.074842,  0] ../../source3/smbd/server.c:1091(smbd_open_one_socket)
Tue Apr 23 10:21:43 2024 daemon.err smbd[18013]:   smbd_open_one_socket: open_socket_in failed: Address not available
Tue Apr 23 10:21:43 2024 kern.warn kernel: [   49.361688] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
Tue Apr 23 10:21:43 2024 kern.warn kernel: [   49.361804] NFSD: Using legacy client tracking operations.
Tue Apr 23 10:21:43 2024 kern.info kernel: [   49.361872] NFSD: starting 10-second grace period (net f0000097)
Tue Apr 23 10:21:43 2024 user.info mwan3track[5980]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2). Current score: 5
Tue Apr 23 10:21:43 2024 daemon.notice rpc.statd[18673]: Version 2.5.4 starting
Tue Apr 23 10:21:43 2024 daemon.warn rpc.statd[18673]: Flags: No-Daemon TI-RPC
Tue Apr 23 10:21:43 2024 daemon.notice rpc.mountd[18674]: Version 2.5.4 starting
Tue Apr 23 10:21:43 2024 daemon.notice rpc.statd[18673]: Initializing NSM state
Tue Apr 23 10:21:43 2024 daemon.warn rpc.statd[18673]: Running as root.  chown /var/lib/nfs to choose different user
Tue Apr 23 10:21:46 2024 user.info mwan3track[5980]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2). Current score: 5
Tue Apr 23 10:21:47 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:47.561362407Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Tue Apr 23 10:21:49 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:49.936973787Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Tue Apr 23 10:21:50 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:50Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Tue Apr 23 10:21:50 2024 daemon.err modprobe: failed to find a module named aufs
Tue Apr 23 10:21:50 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:50.525630143Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Tue Apr 23 10:21:50 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:50.528972870Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Tue Apr 23 10:21:50 2024 daemon.err dockerd[18420]: time="2024-04-23T02:21:50.536886505Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Tue Apr 23 10:21:51 2024 kern.warn kernel: [   56.852504] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Apr 23 10:21:51 2024 kern.warn kernel: [   56.852596] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Apr 23 10:21:51 2024 kern.warn kernel: [   56.852680] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Apr 23 10:21:51 2024 kern.warn kernel: [   56.852759] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Apr 23 10:21:51 2024 daemon.info modprobe: nf_conntrack is already loaded
Tue Apr 23 10:21:51 2024 daemon.info modprobe: nf_conntrack_netlink is already loaded
Tue Apr 23 10:21:51 2024 daemon.info avahi-daemon[9507]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
[... cut ...]
Tue Apr 23 10:31:29 2024 daemon.err tailscaled[10848]: 2024/04/23 02:31:29 control: NetInfo: NetInfo{varies=false hairpin=true ipv6=false ipv6os=true udp=true icmpv4=false derp=#7 portmap= link="" firewallmode="ipt-default"}
Tue Apr 23 10:31:55 2024 daemon.err tailscaled[10848]: 2024/04/23 02:31:55 control: NetInfo: NetInfo{varies=true hairpin=true ipv6=false ipv6os=true udp=true icmpv4=false derp=#7 portmap= link="" firewallmode="ipt-default"}
Tue Apr 23 10:32:21 2024 daemon.err tailscaled[10848]: 2024/04/23 02:32:21 control: NetInfo: NetInfo{varies=false hairpin=true ipv6=false ipv6os=true udp=true icmpv4=false derp=#7 portmap= link="" firewallmode="ipt-default"}

jjm2473, 2024-04-23:
你看看mwan3的存活检测设置的是ping什么地址

wldxx888, 2024-04-24:
> 你看看mwan3的全面检测设置是ping什么地址

默认，没有动过
[image]
[image]

jjm2473, 2024-04-25:
[image]

你确定这些能ping通？要么就全部删除不要检测。