istoreos/istoreos #1520: X86物理机，最近版本反而出现挂载硬盘-1  -4这种情况

Somnr123, 2024-04-25:
[image]
重新挂载恢复正常，一重启问题依旧

Somnr123, 2024-04-25:
延迟挂载30秒也没效果

jjm2473, 2024-04-25:
执行`ls -l /sys/block`，结果发出来看看。

Somnr123, 2024-04-25:
> 执行`ls -l /sys/block`，结果发出来看看。

[image]

jjm2473, 2024-04-25:
执行`cat /etc/config/fstab`看看

jjm2473, 2024-04-25:
你是怎么重新挂载的？

Somnr123, 2024-04-25:
> 你是怎么重新挂载的？

Umount /mnt
Mount /mnt命令

jjm2473, 2024-04-25:
那你自己都指定了挂载点，当然没问题啊。还是看看 /etc/config/fstab

Somnr123, 2024-04-25:
> 执行`cat /etc/config/fstab`看看

[image]

Somnr123, 2024-04-25:
> 那你自己都指定了挂载点，当然没问题啊。还是看看 /etc/config/fstab

不能重启，重启后还是-1  -4

jjm2473, 2024-04-25:
看起来/etc/config/fstab也没问题，那只能推测是开机时/sys还没挂载，硬盘却先挂载了。

你试试umount两个分区以后，执行`/usr/libexec/blockmount.sh`，看是不是能自动挂载到正确位置

Somnr123, 2024-04-25:
> 看起来/etc/config/fstab也没问题，那只能推测是开机时/sys还没挂载，硬盘却先挂载了。
> 
> 
> 
> 你试试umount两个分区以后，执行`/usr/libexec/blockmount.sh`，看是不是能自动挂载到正确位置

卸载一个，重新挂载后还是-1[image]

jjm2473, 2024-04-25:
执行`md5sum /rom/usr/libexec/blockphy.sh /usr/libexec/blockphy.sh`看看

Somnr123, 2024-04-25:
> 执行`md5sum /rom/usr/libexec/blockphy.sh /usr/libexec/blockphy.sh`看看

[image]

jjm2473, 2024-04-25:
执行这段脚本看看：
```bash
testfunca(){
local DISK=sda
local path=`readlink /sys/block/$DISK`
echo "path=$path" >&2
local usb=`echo "$path" | grep -oE '/usb\d+/[^:]+'`
echo "usb=$usb" >&2
    if [ -n "$usb" ]; then
        usb=${usb##*/}
        usb=${usb%%-1}
        echo usb`echo "$usb" | sed 's/[-.]/_/g'`
        return 0
    fi
}
testfunca
```

Somnr123, 2024-04-25:
> 执行这段脚本看看：
>
> ```bash
>
> testfunca(){
>
> local DISK=sda
>
> local path=`readlink /sys/block/$DISK`
>
> echo "path=$path" >&2
>
> local usb=`echo "$path" | grep -oE '/usb\d+/[^:]+'`
>
> echo "usb=$usb" >&2
>
>     if [ -n [...]

[image]

jjm2473, 2024-04-25:
看起来你这个grep命令不正常。执行`echo '../path/usb2/2-2/2-2:0-0/' | grep -oE '/usb\d+/[^:]+'`看看

jjm2473, 2024-04-25:
还有执行`ls -l $(which grep)`看看

jjm2473, 2024-04-25:
哦，我知道了，应该是你安装了grep这个包，覆盖掉了默认的busybox提供的grep，导致结果不一致

jjm2473, 2024-04-25:
你可以先卸载掉grep和sed（如果安装了），或者等下个版本固件发布

Somnr123, 2024-04-25:
> 还有执行`ls -l $(which grep)`看看

我执行了这俩命令，你看下[image]

Somnr123, 2024-04-25:
> 你可以先卸载掉grep和sed（如果安装了），或者等下个版本固件发布

自己没安装过，但是能搜到[image]

jjm2473, 2024-04-26:
可以确定你安装了，固件只会自带busybox版本的。至于怎么安装的我就不知道了。

Somnr123, 2024-04-26:
> 可以确定你安装了，固件只会自带busybox版本的。至于怎么安装的我就不知道了。

这个挂载问题可以解决吗🤣

jjm2473, 2024-04-26:
> 你可以先卸载掉grep和sed（如果安装了），或者等下个版本固件发布

前面不是说了吗？

Somnr123, 2024-04-26:
> > 你可以先卸载掉grep和sed（如果安装了），或者等下个版本固件发布
> 
> 
> 
> 前面不是说了吗？

是在系统-软件包里面搜到，然后卸载删除是吗，上面回我的没看明白

jjm2473, 2024-04-26:
嗯，从软件包里面卸载就行了

jjm2473, 2024-04-26:
卸载以后执行`echo '../path/usb2/2-2/2-2:0-0/' | grep -oE '/usb\d+/[^:]+'`应该结果就不一样了

Somnr123, 2024-04-26:
> 嗯，从软件包里面卸载就行了

删除不掉，会报错[image]

Somnr123, 2024-04-26:
> 卸载以后执行`echo '../path/usb2/2-2/2-2:0-0/' | grep -oE '/usb\d+/[^:]+'`应该结果就不一样了

再搜软件包里没有grep了，应该是卸载了。我执行一下[image]

jjm2473, 2024-04-28:
那暂时没问题了。看起来是“移动通信模块”这个插件依赖了grep。

Somnr123, 2024-04-28:
> 那暂时没问题了。看起来是“移动通信模块”这个插件依赖了grep。

这样是修复好了吗，我现在重启也不会出现挂载-1  -4现象吗？

jjm2473, 2024-04-28:
试试就知道了

jjm2473, 2024-04-28:
至少grep没问题了

Somnr123, 2024-04-28:
> 试试就知道了

您太厉害了，我重启下没有问题，请问下是我安装istore里面软件的问题，还是更新版本的原因，我好规避一下，以防再次出现问题

jjm2473, 2024-04-28:
暂时先别安装“移动通信模块”这个插件就行了。等之后固件或者插件兼容以后再安装。

Somnr123, 2024-04-28:
> 暂时先别安装“移动通信模块”这个插件就行了。等之后固件或者插件兼容以后再安装。

明白了，感谢大佬

closed 2024-04-29 by Somnr123