istoreos/istoreos #1525: plex换了新的512G msata硬盘安装完成打不开主页

qyq3553867, 2024-04-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
之前都是用的128G msata硬盘，可以正常安装运行plex，但是我想想多存几部电影看看，买了一个512G金储星的固态硬盘，结果试了各种办法都是打不开主页，jellyfin这个插件没有问题，128G的和512G的都可以安装运行。
A：

### (2) 路由器型号和固件版本

A：J4125 最新ISTORE OS固件

### (3) 详细日志和/或截图
[image]

A：

qyq3553867, 2024-04-27:
stdout: [migrations] started
stdout: [migrations] no migrations found
stdout: usermod: no changes
stdout: ───────────────────────────────────────
stdout: 
stdout:       ██╗     ███████╗██╗ ██████╗
stdout:       ██║     ██╔════╝██║██╔═══██╗
stdout:       ██║     ███████╗██║██║   ██║
stdout:       ██║     ╚════██║██║██║   ██║
stdout:       ███████╗███████║██║╚██████╔╝
stdout:       ╚══════╝╚══════╝╚═╝ ╚═════╝
stdout: 
stdout:    Brought to you by linuxserver.io
stdout: ───────────────────────────────────────
stdout: 
stdout: To support LSIO projects visit:
stdout: https://www.linuxserver.io/donate/
stdout: 
stdout: ───────────────────────────────────────
stdout: GID/UID
stdout: ───────────────────────────────────────
stdout: 
stdout: User UID:    911
stdout: User GID:    911
stdout: ───────────────────────────────────────
stdout: 
stdout: Change in ownership detected, please be patient while we chown existing files
stdout: This could take some time
stderr: /usr/bin/find: unknown predicate `-R'
stdout: **** Permissions could not be set. This is probably because your volume mounts are remote or read-only. ****
stdout: **** The app may not work properly and we will not provide support for it. ****
stdout: Temporarily starting Plex Media Server.
stdout: Waiting for Plex to generate its config
stderr: PMS: failure detected. Read/write access is required for path: /config/Library/Application Support/Plex Media Server

jjm2473, 2024-04-30:
检查分区是不是只读，文件系统是不是EXT4

closed 2024-04-30 by jjm2473

qyq3553867, 2024-04-30:
> 检查分区是不是只读，文件系统是不是EXT4

分区都对的呀，也是ext4。jellyfin都正常运行的

jjm2473, 2024-04-30:
那就看是不是只读了啊。要么就是你自己设置错了配置文件路径。

```
stderr: /usr/bin/find: unknown predicate `-R'
stdout: **** Permissions could not be set. This is probably because your volume mounts are remote or read-only. ****
stdout: **** The app may not work properly and we will not provide support for it. ****
stdout: Temporarily starting Plex Media Server.
stdout: Waiting for Plex to generate its config
stderr: PMS: failure detected. Read/write access is required for path: /config/Library/Application Support/Plex Media Server
```

qyq3553867, 2024-04-30:
问题是同样的默认设置128G的固态硬盘完全正常，感觉就是个BUG，哪里看只读，问题我一模一样的设置128G的那块固态就没问题，正常运行plex

BarneyShi, 2024-05-02:
> 问题是同样的默认设置128G的固态硬盘完全正常，感觉就是个BUG，哪里看只读，问题我一模一样的设置128G的那块固态就没问题，正常运行plex

我遇到了同样的问题，买了个N100小主机当旁路由，jellyfin和emby都能运行，但是plex就是有这个问题。
请问你最后解决了吗？

qyq3553867, 2024-05-03:
没有啊，还有一个朋友也是这个问题，我估计就是plex插件的BUG了，我512G的金储星硬盘用了各种方法都是不行

BarneyShi, 2024-05-03:
> 没有啊，还有一个朋友也是这个问题，我估计就是plex插件的BUG了，我512G的金储星硬盘用了各种方法都是不行 Barney Shi ***@***.***> 于2024年5月3日周五 03:31写道：
> […](#)

哥们，我终于解决了。首先我换了镜像，不知道怎么回事就突然能进去server，我以为成功了，但是client端不行，是空白。
后来查到是Secure connection这个setting有问题，你需要从preferred改成required，然后再改回来…
棘手的是，我改成required之后又tm进不去了，然后必须要去找到Configs/Plex/preference.xml这个文件，手动把里面的securesetting从0改成1，然后就成功了。
你可以试试，我不知道行不行

qyq3553867, 2024-05-07:
哦。好复杂啊，这个应该也是plex插件的问题？你有TG吗加我下6193205279，这个我128G的固态硬盘没有这个问题啊怎么回事啊