istoreos/istoreos #1526: N1刷不进emmc

drama8888, 2024-04-29:
固件烧录到U盘启动，输入刷进emmc，刷入emmc命令后断电拔U盘，然后上电，只亮logo灯，HDMI无反应，U盘不识别，接网线在路由器里面也看不到ip

jjm2473, 2024-04-30:
hdmi需要开机前就接上。如果还是没反应，那你先把系统刷回官方安卓固件，然后重新刷一遍istoreos。

closed 2024-04-30 by jjm2473

drama8888, 2024-04-30:
> hdmi需要开机前就接上。如果还是没反应，那你先把系统刷回官方安卓固件，然后重新刷一遍istoreos。

大佬你好，hdmi我是开机前就接上的，后续我救砖刷回官方安卓固件又刷了一遍istoreos。还是同样的问题，恩山论坛这个问题我看挺多的，https://www.right.com.cn/forum/thread-8366193-1-2.html。我把帖子地址贴这里，幸苦大佬检查下。

jjm2473, 2024-04-30:
反正我是恢复到官方固件再刷就没问题，如果你们有问题，就寄个测试机过来。注意，官方固件要是2.19以下的，只要刷过linux应该都知道这个。

drama8888, 2024-04-30:
> 反正我是恢复到官方固件再刷就没问题，如果你们有问题，就寄个测试机过来。注意，官方固件要是2.19以下的，只要刷过linux应该都知道这个。

好的，我再反复测试下

sailcom, 2024-09-18:
遇到了同样的问题。
设备是N1，当前是armbian系统。
打算用U盘引导刷入istoreOS，但插上U盘通电开机后，无法进入U盘的istoreOS。
使用了网上提供的以下几种方案都告失败：
1. 更换U盘、换USB2.0、USB3.0、换读卡器+tf小卡；
2. 更换刷盘的软件：rufus、balenaEtcher、Win32DiskImager这几款都尝试过了；
3. 优盘往外拔一点（因为N1的USB口比较深）；
4. 很奇怪的是，U盘里刷armbian可以启动，但istoreOS无法启动（刷入的istoreOS根目录下只有这几个文件）
[image]

jjm2473, 2024-09-18:
@sailcom 那就恢复出厂固件2.19以下，再按教程重来。只有几个文件事正常的，因为这是boot分区。还有很多文件的是rom分区，windows不会挂载的。

sailcom, 2024-09-18:
> @sailcom 那就恢复出厂固件2.19以下，再按教程重来。只有几个文件事正常的，因为这是boot分区。还有很多文件的是rom分区，windows不会挂载的。

谢谢，那我再试试

putianchaoxie, 2025-02-18:
N1盒子后续都没有更新了 估计不会修复了