istoreos/istoreos #1529: istoreos无故断网且无法进入管理页面

Vash90, 2024-04-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
检索了，大部分都能直接进入后台，没有找到相似案列。

### (1) 具体问题

A：突然断网，istoreos的页面无法访问。
但是，接上显示器后能看到 Linux的界面，istores>
linux小白，不知道从合调查，网上说istoreos的日志在 /var/log下面，
但是翻了下也没看到明显的问题，错误报告。

### (2) 路由器型号和固件版本

A：HP T430 瘦客服端，N4000，usb网卡88179
istoreos: istoreos-22.03.6-2024031514-x86-64-squashfs-combined.img.gz
安装了，DS-lite

### (3) 详细日志和/或截图

A：不知道具体要看那个 日志文件
[image]

likaia, 2024-04-30:
我也出这个问题了

Vash90, 2024-04-30:
> 我也出这个问题了
我之前用的r2s,刚换到x86的。之前r2s就算高负债自己卡死都不带断网的。
这x86不知道怎么了。按理说通用性更高呀。低负载也会无故断网，重启恢复。

likaia, 2024-04-30:
> 关键是断网的时候系统还有响应吗？有没有接显示器键盘，断网的时候按键盘回车键有反应吗？

没响应了，我重启了好几次机器，然后我用ssh进后台也进不去，ping ip也不痛。接了hdmi，系统是正常启动了的，也没报错

likaia, 2024-04-30:
> 关键是断网的时候系统还有响应吗？有没有接显示器键盘，断网的时候按键盘回车键有反应吗？

插电脑试了下，lan口不分配ip出来

jjm2473, 2024-04-30:
接键盘啊，按回车看看显示器有没有反应

likaia, 2024-04-30:
> 接键盘啊，按回车看看显示器有没有反应

有反应，启动成功了的，lan口的设备也都显示出来了，但是我在软路由上ping不通

Vash90, 2024-04-30:
> 关键是断网的时候系统还有响应吗？有没有接显示器键盘，断网的时候按键盘回车键有反应吗？

我的有反应，能敲命令，但不知道看 具体 哪个日志。日志路径 对吗？ /var/log/，我体贴了截图。

SunHHB, 2024-04-30:
应该是内存泄漏，看看日志什么应用有内存 out memory。

jjm2473, 2024-04-30:
恢复出厂设置，不要使用翻墙插件和去广告插件，看看还有没有问题。

Vash90, 2024-04-30:
> 恢复出厂设置，不要使用翻墙插件和去广告插件，看看还有没有问题。

感谢老大恢复。能知道具体看哪个日志文件吗？我装系统都是新的，不翻墙没用去广告插件。只装了 迅雷，电子书docker，开启了samba。网络需要，下了 ds-lite。
最早用4月最新固件，发生过。现在用三月的固件也发生过两次。而且完全没有高负载。
目前暂时继续 用的 三月固件。
想知道 下次 再发生  断网无后台的时候，看什么 日志能定位问题？

Vash90, 2024-05-02:
最新觉得问题 可能出在我的 usb网卡  axis88719A 上的。网上有说它驱动不稳定，长时间高负载就断网。或者换到usb2.0端口。
每次断网还有个特点，链接显示器，在Linux后台 reboot 后依然不能联网。必须断电重启才能正常。

jjm2473, 2024-05-06:
那确实有可能是USB网卡的锅。如果要稳定性，就别用USB网卡。

closed 2024-05-06 by Vash90

fredwen2008, 2024-08-14:
最近也出现了几次类似情况。网页无法登录，也不能上网，但是可以ssh到软路由，ssh登录到软路由看起来都正常，从软路由可以访问外网。重启后问题消失，怀疑和passwall有关。

452138222, 2024-11-27:
断网太常见了，电脑连接lan，wan口连接交换机，交换机连接路由，显示没有网络。。 实际是有网的

stofancy, 2025-10-27:
我也是，应该在哪找日志呢？