istoreos/istoreos #1532: 挂载网络存储时，无法选择自定义用户

Asunfly, 2024-04-30:
挂载网络存储时，无法选择自定义用户
[image]
[Powered by LuCI istoreos-22.03 branch (git-24.057.30647-6659faa)](https://github.com/openwrt/luci) / [ArgonTheme v2.2.10.10](https://github.com/jerrykuku/luci-theme-argon) / iStoreOS 22.03.6 2024031514

Vash90, 2024-04-30:
我也是新人啊，不知道猜的对不对。感觉你在 挂载你自己啊？
如果是把 192.168.100.1 istoreos  共享出去给别的电脑访问的话，不是在这个界面。
这个界面是把 其他网络的盘 挂到 你的 istore里。

Asunfly, 2024-04-30:
> 我也是新人啊，不知道猜的对不对。感觉你在 挂载你自己啊？ 如果是把 192.168.100.1 istoreos 共享出去给别的电脑访问的话，不是在这个界面。 这个界面是把 其他网络的盘 挂到 你的 istore里。

第二个填的是nas的路径，哈哈，照样没法选中自己填的用户名、密码

Vash90, 2024-04-30:
其实 我自己用 这个里面的共享。我直接就是用匿名用户，所以没试过 指定用户和密码的情况
[image]

138161, 2024-05-01:
我的好像有啊，admin  guest  自定义
是不是istore版本问题
[Powered by LuCI istoreos-22.03 branch (git-24.057.30647-6659faa)](https://github.com/openwrt/luci)
iStoreOS 22.03.6 2024030112
                                       1514
哈哈，尝鲜了，UI不稳定，有几个版本好像是会有类似的情况，按钮丢失，输入没有等等

jjm2473, 2024-05-08:
@Vash90 他应该问的是挂载，不是共享

jjm2473, 2024-05-08:
@Asunfly 正常情况下有个滚动条，可以填入自定义用户名

[image]

可能你的浏览器有问题，试试chrome。

closed 2024-05-08 by jjm2473