istoreos/istoreos #1535: mtk7922无线网卡问题，ac模式无法设置打开，ax模式很多信道设置了没有wifi

lfsakura0, 2024-05-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mtk7922无线网卡，ac模式无法使用，ax模式48信道后面的好像很少能使用，设置了没有wifi；我用过lean和immortalwrt的固件，均能打开设置ac模式，以及ax模式后面的一些信道选择使用能正常收到wifi

istoreos的有一些地方很好，我挺爱用的，希望作者大大能看看问题，打一下补丁修复一下

### (2) 路由器型号和固件版本

A：x86，n4505，mtk7922无线网卡，istoreos-22.03.6-2024043010-x86-64-squashfs-combined-efi.img

### (3) 详细日志和/或截图

A：这是ac模式设置的样子和日志，就少了一个东西
[image]

[image]

这是ax模式设置的样子和日志，没有wifi出现
[image]
[image]

以下是openwrt.ai固件测试，源码是lean的，immortalwrt的固件之前编译试过，wifi和lean是差不多的
[image]
[image]
[image]

jjm2473, 2024-05-13:
5G频段试试157频道，这个频道不需要雷达回避。

liulangvip, 2024-05-20:
我也是mt7922的，直通istore后，无线里完全启动不了，设置不了ap。是没有驱动的原因吗，驱动里也加了bin文件一样不行，直通给win一切正常，能修复下这款驱动吗。我是极摩客m6主机rz616网卡[image]

[image]

Goodsunny312, 2024-05-22:
我也是mt7922，j1900的x86平台，网上的办法都尝试过了，在添加bin文件到根目录后仅能在客户端模式运行，无法当作ap，ax，ac模式，157频段80mhz都试过了都显示发射功率3dbm无法运行，恳求作者修复该款驱动

jjm2473, 2024-05-23:
好像wifi直通都有问题

closed 2024-05-30 by lfsakura0

Kisoul, 2024-08-25:
> > 好像wifi直通都有问题
> 
> 抱歉，之前比较忙，很少看github，关于7922无线网卡问题，我已经卖了，减去烦恼，买了60块的贝尔刷的op做ap，嘎嘎香。而且和上面那哥们差不多的问题，之前我在pve环境下，无论是istoreos，lede，imm的，直通无线网卡都无法驱动wifi，出问题之前都是正常使用的，是突然无线掉了就再也没正常过了。通过pve官方手册的步骤，屏蔽pve本身的无线驱动后就能用，但不稳定，重启物理机那个命令就会失效。但esxi直通无线网卡就能正常使用，当时op的内核版本最高就是6.1的，都试过，之前也换过机子尝试n4120，n4505，物理机直装均没问题。严重怀疑是pve的驱动或者是内核太新了导致的，只在恩山看到这个贴子内容，和我当时的情况一模一样。但我不是很能理解他表达的意思。
> 
> [已解决]pve直通pcie无线网卡 https://www.right.com.cn/forum/thread-8286014-1-1.html (出处: 恩山无线论坛)
> 
> ![image](https://private-user-images.githubusercontent.com/135017778/333682141-80ee3638-0ca5-4eb8-8bce-30496c3af476.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1OTYwMzQsIm5iZiI6MTcyNDU5NTczNCwicGF0aCI6Ii8xMzUwMTc3NzgvMzMzNjgyMTQxLTgwZWUzNjM4LTBjYTUtNGViOC04YmNlLTMwNDk2YzNhZjQ3Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNVQxNDIyMTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMDQ0YjJiMTJjYzdhM2YwNzc5ODNjMDA4NmM4YzlhNmM1ZGZkMmVmYWZkODYyZDViYWJhZDFmOThkYTc2NDEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9taClCqgfoTtYj-ECpaJVTf50dYcX4o2X5hmKgMoaJA)

怎么屏蔽pve自身的Wi-Fi驱动？网上只有屏蔽显卡驱动的教程