istoreos/istoreos #1537: 能否帮忙编译新的usb网卡驱动 x86

Vash90, 2024-05-02:
关联 #1529 的问题。
原始问题 istoreos无故断网且后台无法访问，接显示器和鼠标键盘能操作linux。
目前自己调查和网上搜索判断的原因是  usb有线网卡  ax88719A 驱动不稳定。
网卡芯片:AX88179     正常
网卡芯片:AX88179A   高负载容易断网，接usb2.0端口也能改善
OPENWRT自带的驱动为 kmod-usb-net-asix-ax88179，感觉是旧的吧。

axis公司提供了最新的驱动，但是 linux小白 真的不会编译这些。
有人能帮忙编译一下，提供我试一下吗？
非常感谢。
正在用的是 istoreos-22.03.6-2024031514-x86-64

axis的官网
https://www.asix.com.tw/en/product/USBEthernet/Super-Speed_USB_Ethernet/AX88179A
驱动下载路径
https://www.asix.com.tw/en/support/download/file/1735?time=1714613463964

jjm2473, 2024-05-06:
不一定是驱动问题，可能本身网卡就不行，例如散热不行，或者其他。

Vash90, 2024-05-06:
> 不一定是驱动问题，可能本身网卡就不行，例如散热不行，或者其他。

我插到 电脑usb3.0上 用这个 usb网卡，高强度网络拷贝 文件 就没问题，也没中断。
目前还是怀疑openwrt的驱动。

jjm2473, 2024-05-06:
买个2.5G的网卡试试呗，便宜的也才六七十块钱

Vash90, 2024-05-06:
> 买个2.5G的网卡试试呗，便宜的也才六七十块钱

买了。路上😄

closed 2024-05-10 by Vash90