istoreos/istoreos #1545: 配置zerotier内网穿透后无法连接iStore商店

RobinLeon, 2024-05-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用zerotier作内网穿透，配置了一些防火墙和接口信息后，无法使用iStore商店连不上，无法进行软件更新和下载，但能ping通istore.linkease.com。之后关闭了所有配置过的防火墙、接口信息和zerotier，仍然无法连接iStore。在PVE虚拟机中新开一个istoreOS的虚拟机也是无法连上iStore商店，配置zerotier过程中没有更改过PVE的设置和防火墙

### (2) 路由器型号和固件版本

A：x86 PVE虚拟机
固件版本：istoreos-22.03.6-2024031514-x86-64-squashfs-combined-efi

### (3) 详细日志和/或截图

A：
[image]
[image]
zerotier配置流程：
# 启用软路由中的zerotier
1. 启用 √
2. 自动允许客户端 NAT √
3. 加入zerotier网络ID

# 网络-防火墙-区域
1. 添加新的区域，如VPN
2. 设置VPN到lan、wan口的转发，入站数据、出站数据、转发 全部接受 开启IP动态伪装
3. lan口的转发区域添加VPN，入站数据、出站数据、转发 全部接受
4. 自定义规则处：添加以下规则
    iptables -I FORWARD -i ztjlhtklsl(zerotier虚拟网卡) -j ACCEPT
    iptables -I FORWARD -o ztjlhtklsl -j ACCEPT
    iptables -t nat -I POSTROUTING -o ztjlhtklsl -j MASQUERADE
    
# 网络-接口
1. 添加新接口，命名为zerotier
2. 协议：选择静态地址，设备：选择zt开头的 IPv4地址填写zerotier分配的局域网IP，子网掩码：255.255.255.0
3. 防火墙设置分配给VPN区域

# 从外网访问路由器的smb
1. 编辑路由器当中的/etc/samba/smb.conf.template
2. 更改bind interfaces only这一项，将其从yes更改为no

jjm2473, 2024-05-06:
可能只是网络慢而已

RobinLeon, 2024-05-06:
> 可能只是网络慢而已
一直打开iStore加载，都无法获取数据

jjm2473, 2024-05-06:
看看浏览器请求，是报错还是连接时间过长

RobinLeon, 2024-05-06:
> 看看浏览器请求，是报错还是连接时间过长

[image]
应该是报错了

RobinLeon, 2024-05-06:
> 看看浏览器请求，是报错还是连接时间过长

[image]
报错了，是网络错误

RobinLeon, 2024-05-06:
[image]

> 看看浏览器请求，是报错还是连接时间过长

[image]
报错了，网络错误

jjm2473, 2024-05-06:
点一下network工具栏，看看具体请求以及响应。

哦，还有一点要注意，浏览器上显示的线上软件列表，是浏览器自己发起请求的，不是让软路由发起请求的。你要检查浏览器能不能访问istore.linkease.com

RobinLeon, 2024-05-06:
> 点一下network工具栏，看看具体请求以及响应。
> 
> 哦，还有一点要注意，浏览器上显示的线上软件列表，是浏览器自己发起请求的，不是让软路由发起请求的。你要检查浏览器能不能访问istore.linkease.com

OK，问题解决了，我以为iStore是软路由发起的请求，我的浏览器是访问不了网站的。我换了其他网络就可以现实软件列表了。感谢

closed 2024-05-06 by jjm2473