istoreos/istoreos #1547: 标题：建议使用阿里云OSS和CDN全球加速优化固件下载速度

huangsen365, 2024-05-05:
标题：建议使用阿里云OSS和CDN全球加速优化固件下载速度

## 详细叙述

### (1) 具体问题

当前iStoreOS的固件下载速度较慢，尤其是在以下两个关键的下载路径：

- 固件下载页面: https://fw.koolcenter.com/iStoreOS/
- 路由器自身的在线升级功能路径: cgi-bin/luci/admin/system/ota

这些下载服务经常出现响应缓慢的情况，影响用户体验。

### (2) 路由器型号和固件版本

此问题与特定型号或版本无关，为全局性能问题。

### (3) 详细日志和/或截图

暂无日志，问题可通过尝试下载固件复现。

## 建议的解决方案

建议 iStoreOS 技术团队考虑使用阿里云的OSS加速和CDN全球加速服务来提升固件下载速度。详细建议如下：

- **OSS加速**：通过阿里云的OSS加速（  https://help.aliyun.com/zh/oss/user-guide/enable-transfer-acceleration  ），使用全球加速Endpoint，即oss-accelerate.aliyuncs.com，可显著提升下载速度并减少服务器负载。
- **CDN全球加速**：如果使用CDN全球加速，建议源站设置为基于阿里云的OSS加速的全球加速Endpoint，以便为国内外用户提供更快的访问速度。

实施这些优化不仅能提高下载速度，还能提升用户的整体体验，吸引更多用户使用iStoreOS。

huangsen365, 2024-05-05:
[image]
经常很容易下载失败

jjm2473, 2024-05-06:
主要是没钱啊，CDN都挺贵的，iStoreOS 现在又没啥盈利点，要是有哪家CDN愿意赞助下倒是可以。国外其实有些网站可以托管下固件（例如sourceforge，ghcr），但是国内估计访问速度也一般，甚至访问不了。

jjm2473, 2024-05-06:
gitee能做文件托管吗？而且固件加起来还不小，目前至少60G了

Vinke2, 2024-05-07:
> gitee能做文件托管吗？而且固件加起来还不小，目前至少60G了

那应该不行

Vinke2, 2024-05-07:
阿里云盘怎么样?

w9315273, 2024-05-10:
```
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4  171M    4 8240k    0     0  14515      0  3:26:02  0:09:41  3:16:21  6574
```

慢到无语 下载个171MB的东西 要3小时 还以为是二十年前呢

jjm2473, 2024-05-11:
@w9315273 不要在高峰期下载就行了。之后可能会加个cdn或者分流站。

w9315273, 2024-05-11:
> 高峰期

我用aria2同时在所有的备用服务器上多线程下载了 不到1分钟

jjm2473, 2024-05-11:
@w9315273 今天单线程下载都很快了。昨天刚发布太多人下载。

jjm2473, 2024-05-13:
@yakult-xia 这个有点风险吧，类似PCDN了，而且估计有些用户也会反感的。直接使用ipfs的话，速度又很慢甚至连不上。之后会弄个cdn和分流站的。

jjm2473, 2024-05-13:
在sourceforge增加了一个分流，可以试试从这里手动下载固件 https://sourceforge.net/projects/istoreos/files/iStoreOS/

jjm2473, 2024-05-13:
CDN要等域名备案以后才能配置好

huangsen365, 2024-05-13:
> 主要是没钱啊，CDN都挺贵的，iStoreOS 现在又没啥盈利点，要是有哪家CDN愿意赞助下倒是可以。国外其实有些网站可以托管下固件（例如sourceforge，ghcr），但是国内估计访问速度也一般，甚至访问不了。

关于赞助事宜，是和你这边对接吗？我这边看看可否给你们撮合争取一下？

jjm2473, 2024-05-13:
@huangsen365 联系这个邮箱 admin@linkease.com ，顺便附上issue链接，这样老板就知道什么事了。你是有阿里云的资源吗？

huangsen365, 2024-05-13:
> @huangsen365 联系这个邮箱 [admin@linkease.com](mailto:admin@linkease.com) ，顺便附上issue链接，这样老板就知道什么事了。你是有阿里云的资源吗？

是的，也的确很巧，因为我们本身需要istores作为办公网络软路由的搭建，在使用的过程中遇到这个苦恼的痛点很困扰，所以一看这个情况就知道没使用CDN造成速度缓慢，接着就尝试提这个issue反馈期望得到改善。感觉istore的的安装使用也很便捷，特别是x86那个版本，不像其它同类的刷固件比较折腾。很看好你们的发展。

后续我这边通过邮件详情和你们老板说说看。

ks-bao, 2024-05-14:
@huangsen365 感谢提供 CDN！搞起来搞起来！其实最近在备案 istoreos.com，之前一直偷懒都没备案。

huangsen365, 2024-05-14:
> @huangsen365 感谢提供 CDN！搞起来搞起来！其实最近在备案 istoreos.com，之前一直偷懒都没备案。
> 
> PS. 貌似还没看到邮件。

刚发送了

ks-bao, 2024-05-15:
> > @huangsen365 感谢提供 CDN！搞起来搞起来！其实最近在备案 istoreos.com，之前一直偷懒都没备案。
> > PS. 貌似还没看到邮件。
> 
> 刚发送了

收到，感谢！istoreos.com 前几天正在备案，这两天应该能备案完成。备案的时候暂时关站了。

huangsen365, 2024-05-15:
> > > @huangsen365 感谢提供 CDN！搞起来搞起来！其实最近在备案 istoreos.com，之前一直偷懒都没备案。
> > > PS. 貌似还没看到邮件。
> > 
> > 
> > 刚发送了
> 
> 收到，感谢！istoreos.com 前几天正在备案，这两天应该能备案完成。备案的时候暂时关站了。

好的，你们在邮件回复提供实体名称，我这边报备一下