istoreos/istoreos #1551: R4S 最新固件 重启路由器 web 页进不去

chouchouxsl, 2024-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

在没启动路由器之前 页面都是可以正常访问的 重启之后web页访问不了报错 当初有通过 istore 安装过IStore x 和 路由狗 不好用让我删除了 还有就是主题加了壁纸 

### (2) 路由器型号和固件版本

设备 R4S 
系统 iStoreOS 22.03.6, 2024043010 

### (3) 详细日志和/或截图

Status: 500 Internal Server Error
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'quickstart/home'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'quickstart/main'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/argon/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: attempt to index local 'boardinfo' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:699: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:699: in function 'include'
	[string "/usr/lib/lua/luci/view/quickstart/main.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	[string "/usr/lib/lua/luci/view/quickstart/home.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:1052: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:482: in function

jjm2473, 2024-05-06:
终端执行`ubus call system board`看看显示什么

chouchouxsl, 2024-05-07:
> 终端执行`ubus call system board`看看显示什么

[image]

我用reboot命令重启的 就会有这个问题  断电重连 就突然好了

closed 2024-05-07 by jjm2473