istoreos/istoreos #1553: IStoreOS 无法开启DHCP服务，或者说开启了，但服务无法生效（小米AX3000T做ap使用）

lxwsdtc88, 2024-05-07:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题
无法开启DHCP服务

Q：是否用关键词搜索了issue? (使用 "x" 选择)
 [x ] 没有类似的issue

 2. 详细叙述
 (1) 具体问题

A：在单位有一个主路由，无DHCP服务，自己设置了一个IStoreOS，开启了DHCP，下面使用小米AX3000T硬路由做ap接入点，发现无法分配ip给设备，主路由 192.168.30.254  istoreos 192.168.30.162  ax3000t 192.168.30.180 dhcp池 192.168.30.120-192.168.30.149，此外ax3000t的ap模式是之前接在apple的airport获取的（这玩意居然可以设置静态ap地址并且自带DHCP服务），当分配了ap后，在istoreos中发现ax3000t的IP后，拆除airport，想用istoreos的dhcp时就如上诉问题，求解答，谢谢

 (2) 路由器型号和固件版本

A：软路由x86 pve 虚拟机IStoreOS22.03.62024030112  ap AX3000T MiWiFi 稳定版 1.0.64

(3) 详细日志和/或截图

A：路由ip设置
[image]
     接口dhcp服务设置
[image]
[image]
    Dnsmasq dhcp服务设置
[image]
  防火墙设置
[image]

lxwsdtc88, 2024-05-08:
昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

ming883, 2024-06-14:
我强制勾选了，还是没用。通过改文件的方式，打开的dhcpv4，就可以了。是bug吗？

jjm2473, 2024-06-17:
默认情况下，如果检测到同网络下有其他dhcp服务就不会启动了。如果已经排除了其他dhcp服务，那么重启下dnsmasq或者重启系统就会开启dhcp服务了。

closed 2024-06-17 by jjm2473

supersnsd, 2024-06-29:
> 昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

这个有用,但路由里没这个选项的开关?

jjm2473, 2024-07-21:
有开关啊，但这是其他插件修改的，是插件bug

tnnevol, 2024-09-24:
> ![image](https://private-user-images.githubusercontent.com/73932839/328571821-8822247d-ae05-4021-8da3-bfd1f1f5e2a3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNDk0NjMsIm5iZiI6MTcyNzE0OTE2MywicGF0aCI6Ii83MzkzMjgzOS8zMjg1NzE4MjEtODgyMjI0N2QtYWUwNS00MDIxLThkYTMtYmZkMWYxZjVlMmEzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDAzMzkyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4YmQ2MjNhNzhkOWRlODkzZjQzZDc5ZTJlODc0ZWFhZTQzZDUxZmUwOTJhOGI5YjM5N2U3OWNlZTQ0MWQyMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Lkkd48MLOedRbhs-_1IjKiPq_A-JIGA_b007vE7SNWE) bro，我刚刚弄好了，我把强制打开就好了，你去试试？

我也是勾选了这个，但是隔离几天后突然又不能分配dhcp服务了

tnnevol, 2024-09-24:
> 默认情况下，如果检测到同网络下有其他dhcp服务就不会启动了。如果已经排除了其他dhcp服务，那么重启下dnsmasq或者重启系统就会开启dhcp服务了。

我目前也是出现了同样问题，我这边设备是r2s 拨号 tplink ax3000 ap模式（这个设备ap模式后dhcp会自动关闭且设置禁用状态），wifi连接后，会有ip 分配到设备，但是很短暂，过一会儿设备的ipv4就没了，只剩下ipv6了，重启r2s 和ax3000后，ip还是短暂刷一下，然后就没了。

jjm2473, 2024-09-24:
1. 检查局域网是否还有其他dhcp服务
2. 看看istoreos的系统日志，dnsmasq有没有报错
3. 执行`ps ww|grep dnsmasq|grep -v grep`看看有没有dnsmasq进程

tnnevol, 2024-09-24:
> 1. 检查局域网是否还有其他dhcp服务
> 2. 看看istoreos的系统日志，dnsmasq有没有报错
> 3. 执行`ps ww|grep dnsmasq|grep -v grep`看看有没有dnsmasq进程

非常感谢

ljm15913101281, 2024-09-27:
昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

这个就可以解决了，这么大一个bug 还新人找半天哈哈

komazhou, 2025-03-20:
> 昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

非常感谢，在尝试了2个小时的设置之后，还好找到了您的这个方法，顺利解决了该问题

situkunhua, 2025-04-25:
> 昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

参考修改了文件，可以，感谢

AlanWang2333, 2025-06-17:
> 昨天翻了一下issues发现可能是因为我开着smartdns和adg导致53端口被重定向了，打开etc/config/dhcp 发现dhcpv4 ‘disabled’，后来直接把 ‘disable’ 改成 ‘server’ ，就可以使用了，因为特殊原因，不想其他接入网络的设备使用istoreos的服务，所以没有勾选强制

我把 	option dhcpv4 'disabled' 这行命令删掉也可以，我觉得 “如果检测到同网络下有其他dhcp服务就不会启动DHCP”是因噎废食，能用总比没得用好，设备到手用不了又没有说明，不知道哪里出错了，就很烦。有多少人能找到GitHub的issue

AlanWang2333, 2025-06-17:
> 默认情况下，如果检测到同网络下有其他dhcp服务就不会启动了。如果已经排除了其他dhcp服务，那么重启下dnsmasq或者重启系统就会开启dhcp服务了。

发现一个问题，如果刷iStoreOS旁路由，铁定会出现这个问题，也没有像你说的那样：重启就就开启DHCP了

hang1233, 2025-06-18:
我也遇到了图形界面dhcp强制开启不生效,  vi  etc/config/dhcp  dhcpv4 ‘disabled’， 把 ‘disable’ 改成 ‘server’ 解决

jjm2473, 2025-06-23:
> 我把 option dhcpv4 'disabled' 这行命令删掉也可以，我觉得 “如果检测到同网络下有其他dhcp服务就不会启动DHCP”是因噎废食，能用总比没得用好，设备到手用不了又没有说明，不知道哪里出错了，就很烦。有多少人能找到GitHub的issue

这个`option dhcpv4 'disabled'`跟“如果检测到同网络下有其他dhcp服务就不会启动DHCP”没有任何关系，`option dhcpv4 'disabled'`是你安装的某些插件改的（例如翻墙/去广告）。

“如果检测到同网络下有其他dhcp服务就不会启动DHCP”本身没问题，因为这说明内网至少有一个DHCP服务了，不至于分配不了IP。

总之，`option dhcpv4 'disabled'`是你安装的某些插件改的（例如翻墙/去广告），你应该检查插件设置。