istoreos/istoreos #1555: > Unable to find partition 1 device, skipped.

tonyshva, 2024-05-07:
> Unable to find partition 1 device, skipped.

你的硬件是什么？开机以后终端执行`cat /tmp/.bootdisk; lsblk; ls -l /dev/`，然后开个issue把结果贴出来。

_Originally posted by @jjm2473 in https://github.com/istoreos/istoreos/discussions/1531#discussioncomment-9323813_

 硬件是 H69k，输出结果如下：

root@iStoreOS:~# cat /tmp/.bootdisk; lsblk; ls -l /dev/
MAJOR=179
MINOR=0
DEVNAME=mmcblk1
DEVTYPE=disk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
nbd0          43:0    0    0B  0 disk 
nbd1          43:32   0    0B  0 disk 
nbd2          43:64   0    0B  0 disk 
nbd3          43:96   0    0B  0 disk 
nbd4          43:128  0    0B  0 disk 
nbd5          43:160  0    0B  0 disk 
nbd6          43:192  0    0B  0 disk 
nbd7          43:224  0    0B  0 disk 
mmcblk1      179:0    0 29.1G  0 disk 
├─mmcblk1p1  179:1    0   64M  0 part /boot
├─mmcblk1p2  179:2    0  256M  0 part /rom
├─mmcblk1p3  179:3    0    2G  0 part /overlay
└─mmcblk1p4  179:4    0 26.7G  0 part /mnt/mmcblk1p4
mmcblk1boot0 179:8    0    4M  1 disk 
mmcblk1boot1 179:16   0    4M  1 disk 
mmcblk0      179:24   0  7.4G  0 disk 
└─mmcblk0p1  179:25   0  7.4G  0 part 
zram0        253:0    0    0B  0 disk 
nbd8          43:256  0    0B  0 disk 
nbd9          43:288  0    0B  0 disk 
nbd10         43:320  0    0B  0 disk 
nbd11         43:352  0    0B  0 disk 
nbd12         43:384  0    0B  0 disk 
nbd13         43:416  0    0B  0 disk 
nbd14         43:448  0    0B  0 disk 
nbd15         43:480  0    0B  0 disk 
crw-------    1 root     root       10,  60 May  3 02:00 ashmem
crw-------    1 root     root       14,   4 May  3 02:00 audio
crw-------    1 root     root       10, 235 May  3 02:00 autofs
crw-------    1 root     root       10, 234 May  3 02:00 btrfs-control
drwxr-xr-x    3 root     root            60 May  3 02:00 bus
crw-------    1 root     root       10,  63 May  3 02:00 cachefiles
crw-------    1 root     root      511,   0 May  3 02:00 cec0
crw-------    1 root     root        5,   1 May  3 02:01 console
crw-------    1 root     root       10,  59 May  3 02:00 cpu_dma_latency
drwxr-xr-x    2 root     root           160 May  3 02:00 dma_heap
drwxr-xr-x    2 root     root           120 May  3 02:00 dri
crw-------    1 root     root       14,   3 May  3 02:00 dsp
drwxr-xr-x    2 root     root           140 May  3 02:00 etherd
crw-rw-rw-    1 root     root        1,   7 May  3 02:00 full
crw-------    1 root     root       10, 229 May  3 02:00 fuse
crw-------    1 root     root      254,   0 May  3 02:00 gpiochip0
crw-------    1 root     root      254,   1 May  3 02:00 gpiochip1
crw-------    1 root     root      254,   2 May  3 02:00 gpiochip2
crw-------    1 root     root      254,   3 May  3 02:00 gpiochip3
crw-------    1 root     root      254,   4 May  3 02:00 gpiochip4
crw-------    1 root     root      254,   5 May  3 02:00 gpiochip5
crw-------    1 root     root       10,  53 May  3 02:00 hdmi_hdcp1x
crw-------    1 root     root       10, 183 May  3 02:00 hwrng
crw-------    1 root     root       89,   0 May  3 02:00 i2c-0
crw-------    1 root     root       89,   5 May  3 02:00 i2c-5
crw-------    1 root     root       89,   6 May  3 02:00 i2c-6
crw-------    1 root     root      248,   0 May  3 02:00 iio:device0
drwxr-xr-x    2 root     root            80 May  3 02:00 input
crw-------    1 root     root        1,  11 May  3 02:00 kmsg
srw-rw-rw-    1 root     root             0 May  3 02:00 log
[... 123 lines cut ...]
crw-rw----    1 root     dialout     4,  58 May  3 02:00 tty58
crw-rw----    1 root     dialout     4,  59 May  3 02:00 tty59
crw-rw----    1 root     dialout     4,   6 May  3 02:00 tty6
crw-rw----    1 root     dialout     4,  60 May  3 02:00 tty60
crw-rw----    1 root     dialout     4,  61 May  3 02:00 tty61
crw-rw----    1 root     dialout     4,  62 May  3 02:00 tty62
crw-rw----    1 root     dialout     4,  63 May  3 02:00 tty63
crw-rw----    1 root     dialout     4,   7 May  3 02:00 tty7
crw-rw----    1 root     dialout     4,   8 May  3 02:00 tty8
crw-rw----    1 root     dialout     4,   9 May  3 02:00 tty9
crw-rw----    1 root     dialout   253,   0 May  3 02:00 ttyFIQ0
crw-rw----    1 root     dialout     5,   3 May  3 02:00 ttyprintk
crw-------    1 root     root       10,  57 May  3 02:00 ubi_ctrl
crw-rw-rw-    1 root     root        1,   9 May  3 02:00 urandom
crw-------    1 root     root        7,   0 May  3 02:00 vcs
crw-------    1 root     root        7,   1 May  3 02:00 vcs1
crw-------    1 root     root        7, 128 May  3 02:00 vcsa
crw-------    1 root     root        7, 129 May  3 02:00 vcsa1
crw-------    1 root     root        7,  64 May  3 02:00 vcsu
crw-------    1 root     root        7,  65 May  3 02:00 vcsu1
crw-------    1 root     root       10,  56 May  3 02:00 vendor_storage
crw-------    1 root     root       10, 130 May  3 02:00 watchdog
crw-------    1 root     root      246,   0 May  3 02:00 watchdog0
crw-rw-rw-    1 root     root        1,   5 May  3 02:00 zero
brw-------    1 root     root      253,   0 May  3 02:00 zram0

jjm2473, 2024-05-08:
执行 `echo  mmcblk1 | grep '^.*[0-9]$'` 看看

tonyshva, 2024-05-08:
结果是 mmcblk1

jjm2473, 2024-05-09:
那这没问题。你先试试修复系统组件吧。https://github.com/istoreos/istoreos/wiki/FAQ

tonyshva, 2024-05-11:
使用插件修复了系统组件后，还是同样的无法升级。
尝试使用下载squashfs镜像上传的方式升级，升级之后还是留在了iStoreOS 22.03.6 2024041213。
记得之前大多数时候都是直接在线升级方式进行的，有过一次下载squashfs来进行升级，最近这次升级到2024041213的时候使用的是下载的squashfs镜像来升级的。

log:
2024-05-11 14:52:08
Running Version: 'iStoreOS 22.03.6 2024041213'
Flashing Version: 'iStoreOS 22.03.6 2024051012'
Sending TERM to remaining processes ...
Sending signal TERM to S99adb-enablemo (10751)
Sending signal TERM to adb (10766)
Sending signal TERM to adb (10781)
Sending signal TERM to ddwebdav (11249)
Sending signal TERM to dockerd (11568)
Sending signal TERM to clash (21316)
Sending signal TERM to openclash_watch (23685)
Sending signal TERM to sleep (23746)
Sending signal TERM to uuplugin (23884)
Sending signal TERM to uuplugin (23931)
Sending signal TERM to oled (8766)
Sending KILL to remaining processes ...
Sending signal KILL to uuplugin (23884)
Sending signal KILL to uuplugin (23931)
Switching to ramdisk...
Performing system upgrade...
Reading partition table from image...
Unable to find partition 1 device, skipped.
Unable to find partition 2 device, skipped.
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/mmcblk1...
Upgrade completed

难道说已经成功了但是版本号没变？是否有办法确认这点。

jjm2473, 2024-05-11:
> Unable to find partition 1 device, skipped.

肯定失败了，这表示没检测到运行中的系统分区，升级程序也不知道把数据写到哪个设备里去。

你执行下`cat /proc/self/mountinfo`，把结果发出来看看。

tonyshva, 2024-05-11:
怎么会突然检测不到系统分区，看磁盘信息没什么问题啊

root@iStoreOS:~# cat /proc/self/mountinfo
17 27 179:2 / /rom rw,relatime - squashfs /dev/root ro
18 27 0:18 / /proc rw,nosuid,nodev,noexec,noatime - proc proc rw
19 27 0:19 / /sys rw,nosuid,nodev,noexec,noatime - sysfs sysfs rw
20 19 0:20 / /sys/fs/cgroup rw,nosuid,nodev,noexec,relatime - cgroup2 cgroup2 rw,nsdelegate
23 27 0:23 / /tmp rw,nosuid,nodev,noatime - tmpfs tmpfs rw
24 27 179:3 / /overlay rw,relatime - ext4 /dev/mmcblk1p3 rw
27 1 0:24 / / rw,noatime - overlay overlayfs:/overlay rw,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work
28 27 0:27 / /mnt rw,relatime shared:1 - tmpfs tmpfs rw,size=4096k
29 27 179:1 / /boot rw,noatime - ext4 /dev/mmcblk1p1 rw
22 27 0:22 / /dev rw,nosuid,noexec,noatime - tmpfs tmpfs rw,size=512k,mode=755
30 22 0:28 / /dev/pts rw,nosuid,noexec,noatime - devpts devpts rw,mode=600,ptmxmode=000
31 19 0:5 / /sys/kernel/debug rw,noatime - debugfs debugfs rw
32 19 0:29 / /sys/fs/bpf rw,nosuid,nodev,noexec,noatime - bpf none rw,mode=700
33 19 0:30 / /sys/fs/pstore rw,noatime - pstore pstore rw
36 28 179:4 / /mnt/mmcblk1p4 rw,relatime shared:2 - ext4 /dev/mmcblk1p4 rw
39 18 0:41 / /proc/fs/nfsd rw,relatime - nfsd nfsd rw

tonyshva, 2024-05-11:
> > Unable to find partition 1 device, skipped.
> 
> 肯定失败了，这表示没检测到运行中的系统分区，升级程序也不知道把数据写到哪个设备里去。
> 
> 你执行下`cat /proc/self/mountinfo`，把结果发出来看看。

怎么会突然检测不到系统分区，看磁盘信息没什么问题啊

root@iStoreOS:~# cat /proc/self/mountinfo
17 27 179:2 / /rom rw,relatime - squashfs /dev/root ro
18 27 0:18 / /proc rw,nosuid,nodev,noexec,noatime - proc proc rw
19 27 0:19 / /sys rw,nosuid,nodev,noexec,noatime - sysfs sysfs rw
20 19 0:20 / /sys/fs/cgroup rw,nosuid,nodev,noexec,relatime - cgroup2 cgroup2 rw,nsdelegate
23 27 0:23 / /tmp rw,nosuid,nodev,noatime - tmpfs tmpfs rw
24 27 179:3 / /overlay rw,relatime - ext4 /dev/mmcblk1p3 rw
27 1 0:24 / / rw,noatime - overlay overlayfs:/overlay rw,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work
28 27 0:27 / /mnt rw,relatime shared:1 - tmpfs tmpfs rw,size=4096k
29 27 179:1 / /boot rw,noatime - ext4 /dev/mmcblk1p1 rw
22 27 0:22 / /dev rw,nosuid,noexec,noatime - tmpfs tmpfs rw,size=512k,mode=755
30 22 0:28 / /dev/pts rw,nosuid,noexec,noatime - devpts devpts rw,mode=600,ptmxmode=000
31 19 0:5 / /sys/kernel/debug rw,noatime - debugfs debugfs rw
32 19 0:29 / /sys/fs/bpf rw,nosuid,nodev,noexec,noatime - bpf none rw,mode=700
33 19 0:30 / /sys/fs/pstore rw,noatime - pstore pstore rw
36 28 179:4 / /mnt/mmcblk1p4 rw,relatime shared:2 - ext4 /dev/mmcblk1p4 rw
39 18 0:41 / /proc/fs/nfsd rw,relatime - nfsd nfsd rw

jjm2473, 2024-05-11:
升级过程中检测不到了，也可能是/dev挂载有问题。目前看起来升级程序前还是正常的。

tonyshva, 2024-05-22:
> 升级过程中检测不到了，也可能是/dev挂载有问题。目前看起来升级程序前还是正常的。

有其他什么方式可以尝试一下吗

tonyshva, 2024-06-16:
> 升级过程中检测不到了，也可能是/dev挂载有问题。目前看起来升级程序前还是正常的。

之后推出的几个版本都试了还是不行，
我是否可以通过emmc刷机方式，只刷入img的方式来升级
就是这个教程里的https://doc.linkease.com/zh/guide/istoreos/install_rockchip_sysupgrade.html#_2-%E8%AE%BE%E5%A4%87%E8%BF%9B%E5%85%A5-loader-%E6%A8%A1%E5%BC%8F

jjm2473, 2024-06-17:
@tonyshva 那你升级时选择不保留配置就行了，但是你的配置肯定就没了。可以先在后台备份下配置。

closed 2024-06-23 by tonyshva