istoreos/istoreos #1558: 千兆口连接小米路由器时速率只有百兆，手动改成千兆后小米路由器无法上网

MCB4QAQ, 2024-05-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [✓ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：千兆口连接小米路由器（CR8808）时发现协商速率只有百兆，手动改成千兆后小米路由器无法上网，百兆时小米路由器IPV6服务不可用

### (2) 路由器型号和固件版本

A：P8600三千兆（22.03.6 2024043010最新），小米CR8808（1.2.20）

### (3) 详细日志和/或截图

A：
[image]
[image]

MCB4QAQ, 2024-05-08:
已经使用系统便利工具开启了IPV6

jjm2473, 2024-05-10:
ipv6跟百兆能有什么关系。换网线试试。

MCB4QAQ, 2024-05-10:
> ipv6跟百兆能有什么关系。换网线试试。

只是因为不仅是百兆，而且ipv6都不生效而已，ipv6确实没啥大关系，但是首要解决为啥小米路由器会是百兆

MCB4QAQ, 2024-05-11:
> 百

问题找到了，错用百兆头了

closed 2024-05-11 by MCB4QAQ

jjm2473, 2024-05-12:
所以说换网线啊