istoreos/istoreos #1561: 最新固件版本，更新后speedtest命令行报错

ksi1110, 2024-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：前几天系统提示有升级，更新后运行speedtest命令行测速，提示失败

### (2) 路由器型号和固件版本

A：
固件版本：
iStoreOS 22.03.6 2024043010 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

x86工控机

### (3) 详细日志和/或截图

A：
[image]

[2024-05-10 08:28:23.573] [error] Configuration - SSL connect error (UnknownException)
[2024-05-10 08:28:23.573] [error] Configuration - Cannot retrieve configuration document (0)
[2024-05-10 08:28:23.583] [error] ConfigurationError - Could not retrieve or read configuration (Configuration)
[2024-05-10 08:28:23.583] [error] ConfigurationError - Could not retrieve or read configuration (Configuration)
[error] Configuration - Could not retrieve or read configuration (ConfigurationError)

jjm2473, 2024-05-10:
可能网络问题

ksi1110, 2024-05-11:
> 可能网络问题

网络正常的，我可以用浏览器打开speedtest.net，dns解析也正常，但是命令行失败，准备重新装一下试试看

ksi1110, 2024-05-11:
重新安装脚本后已解决

closed 2024-05-11 by ksi1110