istoreos/istoreos #1565: WebVirtCloud 在实体主机关机重启之后，其虚拟机无法做到自动重启

benzo970210, 2024-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
WebVirtCloud 在实体主机关机重启之后，其虚拟机无法做到自动重启
A：

### (2) 路由器型号和固件版本
n100 处理器，
iStoreOS 22.03.6 2024043010
A：

### (3) 详细日志和/或截图

A：

benzo970210, 2024-05-10:
已经试过虚拟机页面中的自动开启，但是依旧不成功

yuhua2025, 2024-05-11:
n5105也一样

benzo970210, 2024-05-12:
> n5105也一样

是不是所有机器都一样

soulripper13, 2024-05-12:
我也有同样的问题. N100 和 N5105

soulripper13, 2024-05-15:
> 已经试过虚拟机页面中的自动开启，但是依旧不成功

你解决问题了吗？

benzo970210, 2024-05-16:
> > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> 
> 你解决问题了吗？

> > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> 
> 你解决问题了吗？

meiyou

benzo970210, 2024-05-16:
> > > > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> > > 
> > > 
> > > 你解决问题了吗？
> > 
> > 
> > > > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> > > 
> > > 
> > > 你解决问题了吗？
> > 
> > 
> > meiyou

没有解决，你有解决方案吗

soulripper13, 2024-05-16:
> > > > > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> > > > 
> > > > 
> > > > 你解决问题了吗？
> > > 
> > > 
> > > > > 已经试过虚拟机页面中的自动开启，但是依旧不成功
> > > > 
> > > > 
> > > > 你解决问题了吗？
> > > 
> > > 
> > > meiyou
> 
> 没有解决，你有解决方案吗

还没

soulripper13, 2024-05-24:
我们什么时候可以期待修复？或者有解决办法吗？

soulripper13, 2024-06-21:
https://github.com/istoreos/istoreos/issues/1663

tpc188, 2024-07-19:
[]https://github.com/istoreos/istoreos/issues/1663(url) 根据这个设置了，没有生效是什么原因？是步骤不对吗？

soulripper13, 2024-07-19:
> []#1663 根据这个设置了，没有生效是什么原因？是步骤不对吗？

Check if your container name matches and instance also.

tpc188, 2024-07-19:
[https://github.com/istoreos/istoreos/issues/1663](url)以下这两句命令要怎么用？这两句命令里面start_vm.sh和您开头提到的那个文件名为啥不一样呢？
chmod +x /etc/init.d/start_vm.sh
/etc/init.d/start_vm.sh enable

soulripper13, 2024-07-19:
> [https://github.com/istoreos/istoreos/issues/1663](url)以下这两句命令要怎么用？这两句命令里面start_vm.sh和您开头提到的那个文件名为啥不一样呢？ chmod +x /etc/init.d/start_vm.sh /etc/init.d/start_vm.sh enable

you can name it whatever you want. it is just a script. Oh, now i've noticed the mistake. I've corrected it.

soulripper13, 2024-07-19:
You can ssh in you host machine and type the command see if it boots your instance.

docker exec <container name> virsh start <Instance name>

tpc188, 2024-07-19:
SSH测试这个命令是有效的

tpc188, 2024-07-19:
[image]

tpc188, 2024-07-19:
这是我的文件
[image]

soulripper13, 2024-07-19:
Enable it

soulripper13, 2024-07-19:
Then reboot. And see

tpc188, 2024-07-19:
这是我的文件
[image]

soulripper13, 2024-07-19:
Where did you put the file?

soulripper13, 2024-07-19:
In istoreos root.    /etc/init.d/

tpc188, 2024-07-19:
> In istoreos root. /etc/init.d/

[image]

soulripper13, 2024-07-19:
Any progress?

tpc188, 2024-07-19:
> chmod +x /etc/init.d/start_vm.sh
> /etc/init.d/start_vm.sh enable

[image]
这是不是有问题？

soulripper13, 2024-07-19:
I'll get home and check on my system

tpc188, 2024-07-19:
> It says the file doesn't exist

明明文件就在那里啊

soulripper13, 2024-07-19:
`cd /etc/init.d/`
Ls to see if it there.

Then run `chmod +x start_vm.sh`
Then `start_vm.sh enable`

tpc188, 2024-07-19:
> start_vm.sh enable

[image]

soulripper13, 2024-07-19:
`ls /etc/init.d`

soulripper13, 2024-07-19:
you're running original istoreos?

tpc188, 2024-07-19:
> ls /etc/init.d

[image]

soulripper13, 2024-07-19:
it is there. Then why not found???? Making me crazy :'(

tpc188, 2024-07-19:
> you're running original istoreos?

[image]
这个版本的

tpc188, 2024-07-19:
> it is there. Then why not found???? Making me crazy :'(

> it is there. Then why not found???? Making me crazy :'(
我刚看了今天发布了新版本，我更新看看

tpc188, 2024-07-19:
> > it is there. Then why not found???? Making me crazy :'(
> 
> > it is there. Then why not found???? Making me crazy :'(
> > 我刚看了今天发布了新版本，我更新看看

已更新到最新版，遗憾的是问题依旧没有解决

tpc188, 2024-07-19:
> > > it is there. Then why not found???? Making me crazy :'(
> > 
> > 
> > > it is there. Then why not found???? Making me crazy :'(
> > > 我刚看了今天发布了新版本，我更新看看
> 
> 已更新到最新版，遗憾的是问题依旧没有解决

猜测：istoreos安装好后，进行了docker文件夹迁移，再安装的虚拟机/实例，不知是否有关系？

soulripper13, 2024-07-19:
> istoreos安装好后，进行了docker文件夹迁移，再安装的虚拟机/实例，不知是否有关系

shouldn't matter. Mine also is migrated. Problem is the it says file is missing which is weird. Delete the file and try to create it using the istoreos interface.

tpc188, 2024-07-19:
> > istoreos安装好后，进行了docker文件夹迁移，再安装的虚拟机/实例，不知是否有关系
> 
> shouldn't matter. Mine also is migrated. Problem is the it says file is missing which is weird. Delete the file and try to create it using the istoreos interface.

好的，我试试看

tpc188, 2024-07-19:
> > > istoreos安装好后，进行了docker文件夹迁移，再安装的虚拟机/实例，不知是否有关系
> > 
> > 
> > shouldn't matter. Mine also is migrated. Problem is the it says file is missing which is weird. Delete the file and try to create it using the istoreos interface.
> 
> 好的，我试试看

易有云文件管理内重建的文件，一样提示文件没有找到，真是奇怪了！

DJLSZB, 2024-07-27:
文件格式不对，你应该是windows环境下进行编辑的，要改为Unix才行

soulripper13, 2024-07-27:
> 文件格式不对，你应该是windows环境下进行编辑的，要改为Unix才行

Could be the problem. I did mine on Mac so it worked. Didn't think about that.

DJLSZB, 2024-07-27:
> > 文件格式不对，你应该是windows环境下进行编辑的，要改为Unix才行
> 
> Could be the problem. I did mine on Mac so it worked. Didn't think about that.

这个问题我也是昨天刚解决，我一开始也是在windows下进行编辑的，不行，换成mac就好了

waynewie, 2024-10-02:
我是Intel(R) N100 x86/64  内核5.10.221  固件iStoreOS 22.03.7 2024080210 / LuCI istoreos-22.03 branch git-24.207.28853-3b34de8，istore os商店装的WebVirtCloud，在WebVirtCloud里面装了一个homeassistant，但是不能自启，按照上面的方法试了，最后在执行/etc/init.d/start_vm.sh enable的时候，会提示line 9: syntax error: unexpected newline，这个是什么原因呢？
[image]
[image]
[image]
[image]
[image]
[image]

waynewie, 2024-10-02:
已经搞定了，把‹webvirtcloud>和‹homessistant>的‹和>删掉就可以了