istoreos/istoreos #1567: 机器会无响应，过一段时间恢复

jfgg95, 2024-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：偶尔发生机器卡死的情况，隔一段时间会自己恢复

### (2) 路由器型号和固件版本

A：R86S、iStoreOS 22.03.6 2024043010 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

### (3) 详细日志和/或截图

A：内核日志如下
[309234.002357] mmc0: cqhci: timeout for tag 21
[309234.002365] mmc0: cqhci: ============ CQHCI REGISTER DUMP ===========
[309234.002389] mmc0: cqhci: Caps:      0x000030c0 | Version:  0x00000510
[309234.002414] mmc0: cqhci: Config:    0x00001101 | Control:  0x00000000
[309234.002421] mmc0: cqhci: Int stat:  0x00000000 | Int enab: 0x00000006
[309234.002428] mmc0: cqhci: Int sig:   0x00000006 | Int Coal: 0x00000000
[309234.002435] mmc0: cqhci: TDL base:  0x01cb0000 | TDL up32: 0x00000001
[309234.002442] mmc0: cqhci: Doorbell:  0xffff7f0f | TCN:      0x00000000
[309234.002449] mmc0: cqhci: Dev queue: 0x00000000 | Dev Pend: 0x00200000
[309234.002457] mmc0: cqhci: Task clr:  0x00000000 | SSC1:     0x00010008
[309234.002464] mmc0: cqhci: SSC2:      0x00000001 | DCMD rsp: 0x00000000
[309234.002471] mmc0: cqhci: RED mask:  0xfdf9a080 | TERRI:    0x00000000
[309234.002478] mmc0: cqhci: Resp idx:  0x0000000d | Resp arg: 0x00000000
[309234.002480] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[309234.002504] mmc0: sdhci: Sys addr:  0x00000030 | Version:  0x00001002
[309234.002511] mmc0: sdhci: Blk size:  0x00000200 | Blk cnt:  0x00000000
[309234.002518] mmc0: sdhci: Argument:  0x00018000 | Trn mode: 0x00000023
[309234.002525] mmc0: sdhci: Present:   0x1eff0001 | Host ctl: 0x0000003c
[309234.002532] mmc0: sdhci: Power:     0x0000000b | Blk gap:  0x00000080
[309234.002540] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000007
[309234.002547] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00000000
[309234.002554] mmc0: sdhci: Int enab:  0x02ff4000 | Sig enab: 0x02ff4000
[309234.002561] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[309234.002568] mmc0: sdhci: Caps:      0x546ec881 | Caps_1:   0x80000807
[309234.002575] mmc0: sdhci: Cmd:       0x00000d1a | Max curr: 0x00000000
[309234.002583] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[309234.002590] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[309234.002596] mmc0: sdhci: Host ctl2: 0x0000000d
[309234.002604] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x0000000100c25448
[309234.002606] mmc0: sdhci: ============================================
[309234.002614] mmc0: running CQE recovery
[348635.289596] mmc0: cqhci: timeout for tag 14
[348635.289604] mmc0: cqhci: ============ CQHCI REGISTER DUMP ===========
[348635.289628] mmc0: cqhci: Caps:      0x000030c0 | Version:  0x00000510
[348635.289635] mmc0: cqhci: Config:    0x00001101 | Control:  0x00000000
[348635.289642] mmc0: cqhci: Int stat:  0x00000000 | Int enab: 0x00000006
[348635.289649] mmc0: cqhci: Int sig:   0x00000006 | Int Coal: 0x00000000
[348635.289657] mmc0: cqhci: TDL base:  0x01cb0000 | TDL up32: 0x00000001
[348635.289664] mmc0: cqhci: Doorbell:  0xfdffffff | TCN:      0x00000000
[348635.289671] mmc0: cqhci: Dev queue: 0x00000000 | Dev Pend: 0x00004000
[348635.289678] mmc0: cqhci: Task clr:  0x00000000 | SSC1:     0x00010008
[348635.289685] mmc0: cqhci: SSC2:      0x00000001 | DCMD rsp: 0x00000000
[348635.289692] mmc0: cqhci: RED mask:  0xfdf9a080 | TERRI:    0x00000000
[348635.289700] mmc0: cqhci: Resp idx:  0x0000000d | Resp arg: 0x00000000
[348635.289702] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[348635.289725] mmc0: sdhci: Sys addr:  0x00000072 | Version:  0x00001002
[348635.289732] mmc0: sdhci: Blk size:  0x00000200 | Blk cnt:  0x00000000
[348635.289739] mmc0: sdhci: Argument:  0x00018000 | Trn mode: 0x00000033
[... 390 lines cut ...]
[350506.298843] [  16276]     0 16276      281       11    40960        0             0 uptime
[350506.298845] [  16291]     0 16291     6992      437    94208        0             0 smbd[192.168.3.
[350506.298846] [  16297]     0 16297     6992      437    94208        0             0 smbd[192.168.3.
[350506.298848] [  16309]     0 16309     6992      437    94208        0             0 smbd[192.168.3.
[350506.298849] [  16312]     0 16312      281       12    40960        0             0 dhcpv6.script
[350506.298851] [  16315]     0 16315     6992      437    94208        0             0 smbd[192.168.3.
[350506.298853] [  16323]     0 16323     6992      437    94208        0             0 smbd[192.168.3.
[350506.298854] [  16331]     0 16331     6992      437    94208        0             0 smbd[192.168.3.
[350506.298856] [  16340]     0 16340     6992      437    94208        0             0 smbd[192.168.3.
[350506.298858] [  16350]     0 16350     6992      437    94208        0             0 smbd[192.168.3.
[350506.298859] [  16351]   123 16351      281       11    40960        0             0 ntpd-hotplug
[350506.298861] [  16356]     0 16356     6992      438    94208        0             0 smbd[192.168.3.
[350506.298862] [  16360]     0 16360      331       59    40960        0             0 openclash_watch
[350506.298864] [  16362]     0 16362     6992      437    94208        0             0 smbd
[350506.298865] [  16363]     0 16363     4345      370    69632        0             0 nmbd
[350506.298867] [  16366]     0 16366     6992      437    94208        0             0 smbd
[350506.298868] [  16367]     0 16367      187       12    36864        0             0 askfirst
[350506.298870] [  16372]     0 16372     6992      437    94208        0             0 smbd
[350506.298872] [  16378]     0 16378     6992      437    94208        0             0 smbd
[350506.298873] [  16385]     0 16385     6992      437    94208        0             0 smbd
[350506.298875] [  16389]     0 16389      406       16    36864        0             0 ash
[350506.298876] [  16391]     0 16391     6992      437    94208        0             0 smbd
[350506.298878] [  16392]     0 16392      208       15    40960        0             0 odhcp6c
[350506.298880] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=services,mems_allowed=0,global_oom,task_memcg=/services,task=netdata,pid=14037,uid=0
[350506.298930] Out of memory: Killed process 14037 (netdata) total-vm:37432kB, anon-rss:16528kB, file-rss:0kB, shmem-rss:32kB, UID:0 pgtables:92kB oom_score_adj:1000

jjm2473, 2024-05-13:
看起来是emmc或者sd卡有问题。

jjm2473, 2024-05-13:
而且还有oom，也就是内存不足的问题。

closed 2024-05-13 by jjm2473