istoreos/istoreos #1575: istoreos安装迅雷下载无法运行，求解[抱拳]

jhp1015, 2024-05-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

yeqingyeqing, 2024-05-14:
我也是，x86全新安装istoreos，从商店下载，安装镜像后web打不开，显示404 not found，改成宿主网络也不行，docker根目录已经迁移过了。

jhp1015, 2024-05-14:
是不是官方固件的问题？

h88kmax, 2024-05-14:
我也一样！不知道什么时候可以解决

jhp1015, 2024-05-14:
只能等官方更新了。

soulripper13, 2024-05-15:
use cnk3x/xunlei:3.7.1 not cnk3x/xunlei:latest

soulripper13, 2024-05-15:
> 我也是，x86全新安装istoreos，从商店下载，安装镜像后web打不开，显示404 not found，改成宿主网络也不行，docker根目录已经迁移过了。

use cnk3x/xunlei:3.7.1 not cnk3x/xunlei:latest

Vash90, 2024-05-15:
看看，docker 里面 迅雷的 容器 启动了吗？有时候，服务里的迅雷点了启动，但是docker没启动

h88kmax, 2024-05-15:
都试过了 明显安装的时候 镜像下载不全！

jhp1015, 2024-05-16:
有解决办法吗

jhp1015, 2024-05-16:
这是新的镜像地址吗？

yeqingyeqing, 2024-05-16:
> > 我也是，x86全新安装istoreos，从商店下载，安装镜像后web打不开，显示404 not found，改成宿主网络也不行，docker根目录已经迁移过了。
> 
> use cnk3x/xunlei:3.7.1 not cnk3x/xunlei:latest

谢谢，这个镜像是能用的，目前已运行。 cnk3x/xunlei:3.7.1 ，在自定义处拉取这个镜像。

h88kmax, 2024-05-16:
OK 搞定 谢谢

jhp1015, 2024-05-16:
你这是具体怎么操作的？

miaoxiaohu, 2024-05-18:
[image]

miaoxiaohu, 2024-05-18:
> 你这是具体怎么操作的？

[image]

yczxhcq1, 2024-05-20:
换了3.7.1，提示“当前无法使用此页面”  各位怎么解决的？

jjm2473, 2024-05-23:
这个404只是因为新版本没有自动重定向/web到迅雷首页，实际上自己删掉地址栏的/web就行了

jjm2473, 2024-05-23:
不过插件也会改成不加/web

alex086cn, 2024-05-23:
> ![image](https://private-user-images.githubusercontent.com/37968528/331802717-22f09501-bda1-423e-9c65-244cd776cece.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

我也是困扰了两天，刚才发现用新的镜像地址，不勾选host network可以成功，勾选了就提示DNS

jjm2473, 2024-05-23:
@alex086cn 都说了跟dns没什么关系

closed 2024-05-23 by jjm2473

jhp1015, 2024-05-23:
我的x86 istoreos最新版，安装 4.77-5passwall经常断流，重启一下dnsmasq就恢复了，求解

c36128898, 2024-10-15:
> 插件升级到 1.0.3 即可解决此问题，使用哪个版本的镜像都行。大概十分钟以后就可以在istore看到插件升级按钮了。

我更新到了1.0.3，但是还是打不开管理页面，按上面的换自定义源也不行，请问还有其他解决方式么