istoreos/istoreos #1578: USB网卡2.5G RTL8156 访问高延迟1000ms以上，有遇到相同问题的吗

Vash90, 2024-05-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络访问突然高延迟，1000ms以上。ping 192.168.100.1 也有500ms左右。
      重新插拔，网线会恢复正常延迟。
      最近发生过两次了，郁闷，准备着手调查，也想问下有没有遇到同样问题的。

### (2) 路由器型号和固件版本

A：x86，iStoreOS 22.03.6 2024031514
    usb网卡，rtl8156

### (3) 详细日志和/或截图

A：无

真的郁闷了，之前用ax88179的usb网卡，高负载下容易掉驱动直接断网。
最近刚换了 2.5g的rtl8156，高负载都没问题了，高负载时也没高延迟。
就只是浏览着网页，突然就高延迟，只是单纯高延迟，测试网速都是正常。
重新插拔网线，恢复正常。难受。

单网口的 瘦客服端机器，不得不用usb网卡，也不至于这么拉跨吧，
usb网卡还卖不卖了。

zlphoenix, 2024-06-19:
我用的也是usb网卡，rtl8156B网卡，iperf3 测试的时候发现有个奇怪的现象，iStoreOS 作为server，PC作为Client的时候，客户端上传可以到达2.5GBps，而且稳定运行，但是添加了-R参数测试下载流量一定会导致网卡掉线，而且一启动测试就会掉线。更新了 kmod-usb-net-rtl8152 驱动无效.

Vash90, 2024-08-29:
> 我用的也是usb网卡，rtl8156B网卡，iperf3 测试的时候发现有个奇怪的现象，iStoreOS 作为server，PC作为Client的时候，客户端上传可以到达2.5GBps，而且稳定运行，但是添加了-R参数测试下载流量一定会导致网卡掉线，而且一启动测试就会掉线。更新了 kmod-usb-net-rtl8152 驱动无效.

一定一定 不要用 软件包里的驱动，就这个 kmod-usb-net-rtl8152，和fireware 一起卸载了就好了

Vash90, 2024-08-29:
今天 终于搞好了。 把软件包 安装的 kmod-usb-net-rtl8152 卸载，usb网卡就非常正常了。

closed 2024-08-29 by Vash90