istoreos/istoreos #1582: x86设备，千兆宽带电信下载速度减半，从安装到现在已经2个多月了

a5412398, 2024-05-16:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue：有，但按照说明中的方式尝试，未解决。

## 2. 详细叙述
家中千兆电信，单栈公网IPv4，无IPV6，所有设备IPV6均已关闭，拓扑如下

[image]

### (1) 具体问题

A：迅雷下载到SSD中的速度，最快也就52MB/s，差不多只有带宽一半，家中其他设备下载蓝光电影均可达到100MB/s左右的速度。

### (2) 路由器型号和固件版本

A：
iStoreOS 22.03.6 2024041213 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

防火墙内路由/NAT 分载均已开启。

[image]

数据包引导也打开了。
[image]

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]
[image]

silentxrture, 2024-07-03:
一样

a5412398, 2024-07-05:
> 一样

困扰了我好久，他们说换个固件可以解决...换固件又要挨个设置半天，烦得很

jjm2473, 2024-07-08:
会不会只是因为你的usb3.0硬盘降级到2.0了？52MB/s差不多就是usb2.0的理论极限。

执行`lsusb -tv`看看

a5412398, 2024-07-29:
> 会不会只是因为你的usb3.0硬盘降级到2.0了？52MB/s差不多就是usb2.0的理论极限。
> 
> 执行`lsusb -tv`看看

root@iStoreOS:~# lsusb -tv
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
    ID 1d6b:0003
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/4p, 5000M
        ID 0424:5534
        |__ Port 4: Dev 3, If 0, Class=, Driver=uas, 5000M
            ID 152d:0576
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    ID 1d6b:0002
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/4p, 480M
        ID 0424:2134
    |__ Port 3: Dev 3, If 0, Class=, Driver=hub/4p, 480M
        ID 0424:2514

jjm2473, 2024-07-29:
USB倒是3.0，没降级。FullconeNAT改成高性能模式试试。

a5412398, 2024-07-30:
> USB倒是3.0，没降级。FullconeNAT改成高性能模式试试。

[image]
改了，但是好像还是没啥作用，速度依旧在52MB/S以下，就40M左右徘徊，不管下啥资源都是这样。

jjm2473, 2024-07-30:
那找迅雷反馈吧，也许docker版迅雷有限速

a5412398, 2024-08-01:
> 那找迅雷反馈吧，也许docker版迅雷有限速

换了Aria2和qBittorrent   一样的，包括网关改为旁路由的设备，下载速度全部被限制在52M左右。。换成主路由就能直接满速。看来我还是换个系统吧

lnbiuc, 2025-01-21:
一样

a5412398, 2025-01-21:
> 一样

我重装了解决了,速度能拉满~