istoreos/istoreos #1586: 管理界面登录报错

rikkynice, 2024-05-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：报错信息
/usr/lib/lua/luci/controller/ubuntu.lua:3: module 'luci.model.docker' not found:
	no field package.preload['luci.model.docker']
	no file './luci/model/docker.lua'
	no file '/usr/share/lua/luci/model/docker.lua'
	no file '/usr/share/lua/luci/model/docker/init.lua'
	no file '/usr/lib/lua/luci/model/docker.lua'
	no file '/usr/lib/lua/luci/model/docker/init.lua'
	no file './luci/model/docker.so'
	no file '/usr/lib/lua/luci/model/docker.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/ubuntu.lua:3: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:434: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:456: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:190: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:97: in function </usr/lib/lua/luci/dispatcher.lua:96>

### (2) 路由器型号和固件版本

A：ax9000

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-05-23:
ubuntu插件导致的

gugugu-J, 2024-11-26:
该怎么解决呀，小白不知道如何操作了。现在进不去后台界面了