istoreos/istoreos #1590: 商店里的迅雷安装不上了。

alex086cn, 2024-05-22:
docker pull registry.cn-shenzhen.aliyuncs.com/cnk3x/xunlei:latest
latest: Pulling from cnk3x/xunlei
ea6255c5eee6: Pull complete 
b310612c2302: Pull complete 
Digest: sha256:5ee63e711d1d668d6c869bf2459eccf0cdbfd80e3a3b3cdaa0bbf85fe39c418a
Status: Downloaded newer image for registry.cn-shenzhen.aliyuncs.com/cnk3x/xunlei:latest
registry.cn-shenzhen.aliyuncs.com/cnk3x/xunlei:latest
Error: No such container: xunlei
docker run --restart=unless-stopped -d -h iStoreOS     -v "/mnt/sata1-1/Configs/Xunlei:/xunlei/data"     -v "/mnt/sata1-1/Public/Downloads:/xunlei/downloads" --privileged     --dns=127.0.0.1     --network=host  -e TZ=Asia/Shanghai --name xunlei "registry.cn-shenzhen.aliyuncs.com/cnk3x/xunlei:latest"
WARNING: Localhost DNS setting (--dns=127.0.0.1) may fail in containers.
802521fa62d992f679b43740b52b4e72c95dd6eee3f5520de43151005912cb24

有一项DNS提示。最新的0510版本。

h88kmax, 2024-05-23:
这个发过贴的啊 你把镜像用自定义 cnk3x/xunlei:3.7.1就可以了

alex086cn, 2024-05-23:
依然不行，一样的提示，求测试发自我的 iPhone

h88kmax, 2024-05-23:
你把原来dock里面的迅雷都删除 在把迅雷下载卸载 重启后在安装在试 我用的好好的

alex086cn, 2024-05-23:
> 你把原来dock里面的迅雷都删除 在把迅雷下载卸载 重启后在安装在试 我用的好好的

我之前还用的好好的，就是换了个引导盘，重做了下0510版本，你说的方法我都试过，DOCER里面删除容器，删除镜像，删除卷标，用另外的源下载安装，依然提示一个DNS错误，迅雷也显示运行，但是打开地址访问就是404

jjm2473, 2024-05-23:
这个DNS是正常的，因为`--network=host`，本来就是要使用本地的dns，忽略这个警告

closed 2024-05-23 by jjm2473

lff110, 2024-07-03:
我现在用默认的也能正常下载了

jjm2473, 2024-07-17:
dns本来就只是个警告，又不是错误