istoreos/istoreos #1592: 升级系统后,无法进入luci .ERR_CONNECTION_REFUSED

jg3615207, 2024-05-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
升级系统后,SSH正常,Docker也正常.就是无法进入GUI,出現ERR_CONNECTION_REFUSED
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86  iStoreOS 22.03.6, 2024051012

### (3) 详细日志和/或截图

A：

jg3615207, 2024-05-23:
附Update log

2024-05-23 13:28:51
Running Version: 'iStoreOS 22.03.6 2024031514 oversea'
Sending TERM to remaining processes ...
Sending signal TERM to tailscaled (11549)
Sending signal TERM to S99adb-enablemo (15346)
Sending signal TERM to adb (15355)
Sending signal TERM to adb (15373)
Sending signal TERM to aria2.main (15503)
Sending signal TERM to aria2c (15553)
Sending signal TERM to dockerd (15720)
Sending signal TERM to containerd (16815)
Sending signal TERM to docker-proxy (17491)
Sending signal TERM to docker-proxy (17506)
Sending signal TERM to docker-proxy (17514)
Sending signal TERM to docker-proxy (17521)
Sending signal TERM to docker-proxy (17633)
Sending signal TERM to docker-proxy (17641)
Sending signal TERM to containerd-shim (17692)
Sending signal TERM to containerd-shim (17693)
Sending signal TERM to containerd-shim (17696)
Sending signal TERM to containerd-shim (17698)
Sending signal TERM to sh (17854)
Sending signal TERM to foreground (17855)
Sending signal TERM to apache2 (17880)
Sending signal TERM to s6-svscan (17883)
Sending signal TERM to s6-supervise (18614)
Sending signal TERM to s6-linux-init-s (18617)
Sending signal TERM to sh (19023)
Sending signal TERM to uwsgi (22654)
Sending signal TERM to containerd-shim (5770)
Sending signal TERM to entrypoint.sh (5790)
Sending signal TERM to alist (5813)
Sending signal TERM to busybox-extras (5891)
Sending signal TERM to nginx (6005)
Sending signal TERM to nginx (6006)
Sending signal TERM to nginx (6007)
Sending signal TERM to uwsgi (9709)
Sending KILL to remaining processes ...
Sending signal KILL to dockerd (15720)
Sending signal KILL to sh (17854)
Switching to ramdisk...
Performing system upgrade...
Extract boot sector from the image
Reading partition table from image...
Writing image to /dev/sda1...
Writing image to /dev/sda2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/sda...
Upgrading bootloader on /dev/sda...
Upgrade completed

jjm2473, 2024-05-24:
你是不是从wan口访问的？

jg3615207, 2024-05-24:
不是，我是pve 裝的。更新前都沒問題

tanyf, 2024-05-27:
同样有这个问题。作为虚拟机装在群晖的VMM里。更新后SSH连接没问题，Passwall2代理也正常。就是无法进入LUCI

tanyf, 2024-05-27:
#863 我的问题解决了，楼主可以试一下

jg3615207, 2024-05-27:
> #863 我的问题解决了，楼主可以试一下

升级nginx完就:

/usr/lib/lua/luci/dispatcher.lua:430: /etc/config/luci seems to be corrupt, unable to find section 'main'

kiki-theone, 2024-05-27:
同问，没有解决哦

jjm2473, 2024-05-31:
@jg3615207估计/etc/config/luci文件损坏了

kiki-theone, 2024-06-05:
> #863 我的问题解决了，楼主可以试一下

实测，可以解决我的问题。