istoreos/istoreos #1594: r4s运行一段时间lan口千兆变为百兆

Yvespanda1988, 2024-05-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r4s运行一段时间后lan口协议由千兆变为百兆，需要重刷系统或者拔插网线恢复！

### (2) 路由器型号和固件版本

A：R4s

### (3) 详细日志和/或截图

A：

Mioscout, 2024-06-15:
R2C有同样的问题
并且尝试交换ETH1/0无效，交换过去直接就是百兆
千兆变百兆的时候会断网

luxsa, 2024-08-18:
hk28也有相同问题
[image]

Mioscout, 2024-08-18:
> h28k也有相同问题 ![IMG_9876](https://private-user-images.githubusercontent.com/73064218/358899511-d7125c8d-5420-4db9-837d-4c02143987b9.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM5ODk4NDAsIm5iZiI6MTcyMzk4OTU0MCwicGF0aCI6Ii83MzA2NDIxOC8zNTg4OTk1MTEtZDcxMjVjOGQtNTQyMC00ZGI5LTgzN2QtNGMwMjE0Mzk4N2I5LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxOFQxMzU5MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YTk1ZTgxODE2NzVmYjk1YjJiNzBlODcyNTVhMmExMWRmZTAxZjQxYWUwOTE5YWJiNzYzMzllMmZmM2Y5YjJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LMzPNK5-GulfhLDyZkBaycTFKmMoNRU1IsqN8mfN2GA)

我用最新版本的Istore没问题了

luxsa, 2024-08-18:
> 我用最新版本的Istore没问题了
我的也是最新版，有问题
[image]