istoreos/istoreos #1598: istoreos非常好用，不过docker模块有点无从下手的感觉，请问在哪里可以找到相关的教程

wqinf, 2024-05-24:
istoreos非常好用，不过docker模块有点无从下手的感觉，请问在哪里可以找到相关的教程？
比如每个菜单都有什么用？下载的镜像如何能直接点两下就能实现部署？我找了很久都没找到，最好能有一个应用的部署案例，现在有点无从下手的感觉。
目前在用的极空间NAS的Docker虽然没有特权模式，但是可以很方便的下载镜像和实现部署，点两下就好了，还是很方便的。

希望能指导一下，非常感谢！

138161, 2024-05-25:
https://hub.docker.com/search?q=
目前就用两三个，迅雷，qb，同步
cnk3x/xunlei
johngong/qbittorrent
syncthing

138161, 2024-05-25:
最近增加了 ACME证书，smartDNS, uhttpd ,  alist, gowebdav
其他的实用性就。。。。

138161, 2024-05-25:
qBittorrent-qBittorrentEE
4.6.4-4.6.4.10  (amd64;arm64v8;arm32v7) 集成Trackers自动更新
https://hub.docker.com/r/johngong/qbittorrent
http://192.168.100.1:8989

https://cf.trackerslist.com/all.txt

docker pull johngong/qbittorrent
docker create \
    --name=qbittorrent \
    -e QB_WEBUI_PORT=8989 \
    -e QB_EE_BIN=true \
    -e UID=1000 \
    -e GID=1000 \
    -e UMASK=022 \
    -e TZ=Asia/Shanghai
    --net=host \
    -e QB_TRACKERS_UPDATE_AUTO=true
    -p 6884:6884 \
    -p 6884:6884/udp \
    -p 8989:8989 \
    -v /mnt/sata2-4/Configs:/config \
    -v /mnt/sata2-4:/Downloads \
    --restart unless-stopped \
    johngong/qbittorrent:latest

/config和/downloads在weiUI中不要更改，指相对目录变量

编辑--日志内查看用户名admin和密码uHHP79AWY(每次启动不一样)
缺省的用户名和密码: 

可在此菜单更改： 'Tools' menu -> 'Options...' -> 'Web UI' tab -> 'Authentication'
webUI 白名单  输入IP地址：0.0.0.0/0
端口改为 6884
防火墙自定义规则
# qBittorrent Docker
ip6tables -A INPUT -p tcp    --dport 6884  -j ACCEPT
ip6tables -A INPUT -p udp  --dport 6884  -j ACCEPT

http://1337.abcvg.info:80/announce
http://207.241.226.111:6969/announce
http://207.241.231.226:6969/announce
http://46.17.46.112:8080/announce
http://[2a00:b700:1::3:1dc]:8080/announce
http://[2a01:4f8:c012:8025::1]:8080/announce
http://[2a04:ac00:1:3dd8::1:2710]:2710/announce
http://aboutbeautifulgallopinghorsesinthegreenpasture.online:80/announce
http://bittorrent-tracker.e-n-c-r-y-p-t.net:1337/announce
http://bt.okmp3.ru:2710/announce
http://bvarf.tracker.sh:2086/announce
http://ch3oh.ru:6969/announce
http://ipv6.rer.lol:6969/announce
http://movies.zsw.ca:6969/announce
http://nyaa.tracker.wf:7777/announce
http://open.acgnxtracker.com:80/announce
http://retracker.hotplug.ru:2710/announce
http://retracker.seven-sky.net:80/announce
http://t.acg.rip:6699/announce
http://t.overflow.biz:6969/announce
http://tk.greedland.net:80/announce
http://torrentsmd.com:8080/announce
http://tracker.bt4g.com:2095/announce
http://tracker.dler.com:6969/announce
http://tracker.dler.org:6969/announce
http://tracker.edkj.club:6969/announce
http://tracker.electro-torrent.pl:80/announce
http://tracker.files.fm:6969/announce
http://tracker.gbitt.info:80/announce
http://tracker.ipv6tracker.org:80/announce
http://tracker.ipv6tracker.ru:80/announce
[... 116 lines cut ...]
udp://tracker.moeking.me:6969/announce
udp://tracker.opentrackr.org:1337/announce
udp://tracker.picotorrent.one:6969/announce
udp://tracker.qu.ax:6969/announce
udp://tracker.skyts.net:6969/announce
udp://tracker.srv00.com:6969/announce
udp://tracker.t-rb.org:6969/announce
udp://tracker.therarbg.com:6969/announce
udp://tracker.therarbg.to:6969/announce
udp://tracker.tiny-vps.com:6969/announce
udp://tracker.torrent.eu.org:451/announce
udp://tracker.torrust-demo.com:6969/announce
udp://tracker.tryhackx.org:6969/announce
udp://tracker1.bt.moack.co.kr:80/announce
udp://tracker1.myporn.club:9337/announce
udp://tracker2.dler.com:80/announce
udp://tracker2.dler.org:80/announce
udp://tracker2.itzmx.com:6961/announce
udp://tracker3.itzmx.com:6961/announce
udp://tracker4.itzmx.com:2710/announce
udp://ttk2.nbaonlineservice.com:6969/announce
udp://u4.trakx.crim.ist:1337/announce
udp://u6.trakx.crim.ist:1337/announce
udp://wepzone.net:6969/announce
udp://www.torrent.eu.org:451/announce

wqinf, 2024-05-27:
> https://hub.docker.com/search?q= 目前就用两三个，迅雷，qb，同步 cnk3x/xunlei johngong/qbittorrent syncthing

感谢分享。

closed 2024-06-03 by wqinf