istoreos/istoreos #1599: 3月后的版本ipv4上行只有10兆

kwokzito, 2024-05-24:
安装过几个版本，发现ipv4上行只有10兆，但是ipv6却是正常跑慢上行40兆，Intel(R) Celeron(R) CPU J3455 双网口

jjm2473, 2024-07-08:
翻墙了吧

kwokzito, 2024-07-08:
你好，郭子涛已收到邮件！

kwokzito, 2024-07-08:
> 翻墙了吧

没有呢，都是刚安装完就测试

jjm2473, 2024-07-17:
老版本ipv4上传就没问题？你确定不是测速节点有异常？

kwokzito, 2024-07-17:
> 老版本ipv4上传就没问题？你确定不是测速节点有异常？

肯定都排除了才到这里反馈，目前使用版本[Powered by LuCI openwrt-23.05 branch (git-752be18)]/ [ArgonTheme vgit-25.026.72358-0a2aa30]/ OpenWrt 03.13.2024 by Kiddin'

上传没有经过任何节点，都是直连下载文件测速

jjm2473, 2024-07-17:
如果你是测试外网访问内网文件，那么试试关掉防火墙的flow offload，看速度是不是正常，至少以前有关于flow offload导致内网上行变慢的讨论 https://forum.openwrt.org/t/port-forwarding-is-very-slow-only-when-outbound/37740/13?u=jjm2473 。

kwokzito, 2024-07-17:
> 你不是测上行吗，怎么是下载文件来测的，难道是外网访问内网文件？
> 
> 还有，`[Powered by LuCI openwrt-23.05 branch (git-752be18)]/ [ArgonTheme vgit-25.026.72358-0a2aa30]/ OpenWrt 03.13.2024 by Kiddin`这东西是istoreos？来搞笑的吗？你看看正常的istoreos会有‘Kiddin’这个文字不。 要么你用的根本不是istoreos，要么你自己乱安装其他固件的系统软件包把系统文件都换成其他固件的了。

算了，好心遇到问题想着来反馈，说我搞笑，好人不好当

zybsai, 2025-08-03:
> 如果你是测试外网访问内网文件，那么试试关掉防火墙的flow offload，看速度是不是正常，至少以前有关于flow offload导致内网上行变慢的讨论 https://forum.openwrt.org/t/port-forwarding-is-very-slow-only-when-outbound/37740/13?u=jjm2473 。

请问一下istoreos里面怎么关掉呀,没找到开关呢

kwokzito, 2025-08-03:
你好，已收到邮件！

closed 2025-08-05 by jjm2473