istoreos/istoreos #1603: led灯亮度调节

pili-book, 2024-05-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：感觉客厅的路由led灯太亮，有没有插件或者方法能调节灯的亮度了

### (2) 路由器型号和固件版本

A：r6s

### (3) 详细日志和/或截图

A：

jjm2473, 2024-05-31:
不是所有led都是pwm控制的，软件根本没法调亮度

pili-book, 2024-06-08:
> pwm控制

这个不能软实现吗，必须配合硬件吗，板子本身的频率不行吗

6ang996, 2024-06-08:
sys_led对应的GPIO1_C1不支持pwm，wan(GPIO1_C2)、lan1(GPIO1_C3)、lan2(GPIO1_C4)看起来支持PWM

pili-book, 2024-06-08:
我试了试自定义闪烁间隔模仿pwm 好像亮度调不下去，最低毫秒的级别好像不够？反正是亮度调的不太明显，只能说是能调到每几秒忽闪一下子，但是这种情况吧就没法判断网线是否连接好了。。。。哈，估计不太好实现

6ang996, 2024-06-08:
闪烁间隔肯定不够，可以把sysled关闭，然后其他3路led修改dts配置成pwm模式。或者搞个定时任务晚上把led关闭，白天开启。还可以再发散下，搞个光线传感器，关灯时把led关闭，开灯时开启led

pili-book, 2024-06-08:
这玩意。。我这水平。。。。实在不行，要是官方出个插件还行😥

jjm2473, 2024-06-17:
如果只是嫌led晃眼，关掉就行了。后台的系统-LED，添加一个led配置，选择要关闭的led，将触发器设置成off。

pili-book, 2024-06-17:
这个目前就是这样用的。。总归是还想显示，还想亮度很低是最好的。。。。毕竟重启或者有故障了，就比较麻烦，看楼上有回复说能实现软pwm亮度调节。。。也不知道能不能行。。。。。当然有空的话希望官方琢磨下

jjm2473, 2024-06-17:
找友善官方解决吧，他们要是改好了dts我直接抄过来就是了。

realchris007, 2024-06-19:
> 如果只是嫌led晃眼，关掉就行了。后台的系统-LED，添加一个led配置，选择要关闭的led，将触发器设置成off。

大佬，可否細說一下？小白剛入門幾天，我也是快被晚上的LED燈閃瞎，好苦

pili-book, 2024-06-19:
在系统里有led设置啊，有好几个模式你去看看，设置不复杂