istoreos/istoreos #1604: 文件管理报错：AxiosError: Request failed with status code 500

zorro0109, 2024-05-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首页点击文件管理，报500错误，官网未找到其他查看日志的办法。

### (2) 路由器型号和固件版本

A：
iStoreOS 22.03.6 2024052410

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-05-31:
什么浏览器

zorro0109, 2024-05-31:
chrome

kylin0329, 2024-06-01:
我PVE安装的虚拟机进去之后也是报这个错

yuhua2025, 2024-06-02:
我的问题是：打开文件管理后要等大慨40秒才出现内容
[image]

seafeee, 2024-06-07:
我也是同样的问题。 软路由：r66s 使用的chrome浏览器

seafeee, 2024-06-08:
[image]
@zorro0109 这个功能是易有云提供的，所以得把这个服务打开

closed 2024-06-10 by zorro0109

jjm2473, 2024-06-17:
这个后台API确实是易有云提供的