istoreos/istoreos #1605: 通过 openvpn client 连上 vpn, 访问 Google.com 报 证书错误 - your connection to site is not secure

Grrison, 2024-05-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
*  没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装了 openvpn-openssl 和 luci-app-openvpn, 配置了 openvpn 客户端，可以连上网，但是 不是所有的外网都可以访问， 
访问 google.com 和 youtube.com的时候 报证书 错误 “your connection to this site is not secure”.  查看了一下 证书，用的是 *.instagram.com的证书？ 是不是 固件是 有 默认证书 错了？ 电脑是通过三级路由的无线点接入网络的

相同的电脑，换个 无线接入点，在电脑上通过openvpn client连，没有问题，应该不是电脑的问题，感觉像istoreos里有一批默认证书 

### (2) 路由器型号和固件版本

A：nanopi R5S,  固件：istoreos 22.03.6 2024051012

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-05-31:
这肯定是被攻击了，而且证书跟istoreos没任何关系，这个证书正常情况下是服务器发过来的，客户端进行验证。服务器证书可能会被劫持，在中国访问国外网站这很常见。

jjm2473, 2024-05-31:
还有个可能是你使用了错误的ip或dns，连的服务器根本就不是谷歌

closed 2024-05-31 by Grrison