istoreos/istoreos #1606: 请教下双wan口，静态路由和规则设定，策略路由

Vash90, 2024-05-26:
花了两天时间，在网上查资料也没搞定，来求助下。先感谢各位了
设备：x86 istoreos
需求：指定内网ip，192.168.100.115，通过指定接口 transix 上网。

背景：现在有，wan和transix 两个接口可以连接网络。
wan口跃点 42，transix 跃点 45. 所以，默认通过wan口上网。 改transix的跃点，transix也能联网。
保留wan口跃点更低，默认联网。
希望增加，静态路由和规则，指定192.168.100.115，指定通过 transix联网。
不管我怎么设置 静态路由和规则，都没有成功。 设置取消transix的默认路由也不行。

wan口 和 transix口
[image]
route -n
[image]

路由规则，静态路由表
[image]

istoreos 路由界面
[image]
[image]

jjm2473, 2024-06-07:
transix接口的高级设置选择你要覆盖的路由表，例如填一个新的1001路由表，transix的默认网关就会写到1001路由表，然后在“网络”-“路由”里面，通过路由规则，将你的192.168.100.115关联到新的路由表1001，这样就能实现策略路由。

然后你的ipv4规则里面应该不需要选择传出接口，反正路由以后自然会按下一跳地址选择接口的。

jjm2473, 2024-06-07:
transix要勾选默认网关，只是需要设置下路由表，不能使用默认路由表。

jjm2473, 2024-06-07:
https://man7.org/linux/man-pages/man8/ip-rule.8.html oif肯定是不能设置的，也就是规则中的“传出接口”必须为空。

jjm2473, 2024-06-07:
![](https://private-user-images.githubusercontent.com/6929525/333850168-d79b7f8c-5c74-4de1-b100-c3d45fa7a9d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc3MzU1MDMsIm5iZiI6MTcxNzczNTIwMywicGF0aCI6Ii82OTI5NTI1LzMzMzg1MDE2OC1kNzliN2Y4Yy01Yzc0LTRkZTEtYjEwMC1jM2Q0NWZhN2E5ZDkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDdUMDQ0MDAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NThhZWEyNDljZjFhNTIzZmNhZmJhYmQ1NDUyODAyYmEwZjc1NWRjNjg2YmIyZGYzYTVjNjY1NDE0NDJhZjI5ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.0HmlfMzM9dcm6MivoZaIqHxgnFV3xwAzxNMpQ5ttbzs)

这里设置肯定有问题，网关要么写正确，要么别配置这个路由，而是transix接口勾选默认网关，并设置覆盖路由表252

jjm2473, 2024-06-07:
还有需要注意，别跟mwan3一起用（卸载掉），到时候不知道哪个规则有效，可能还有冲突。

logicrhythm, 2024-06-07:
也可以用mwan3设置策略，相对而言可能简单，不过不足就是方法略显小白，另外首页流量统计那里没法用了~~

jjm2473, 2024-06-07:
### 举几个简单的策略路由例子

注意：以下都是`/etc/config/network`文件的片段

#### 双wan双lan范例：
```ini
config device
        option type 'bridge'
        option name 'br-lan2'
        list ports 'eth3'

config interface 'lan2'
        option proto 'static'
        option netmask '255.255.255.0'
        option ipaddr '192.168.102.1'
        option device 'br-lan2'
        option defaultroute '0'

config interface 'wan2'
        option proto 'dhcp'
        option device 'eth1'
        option ip4table '1001'
        option ip6table '1001'

config rule
        option in 'lan2'
        option lookup '1001'

```
其中的重点是wan2的`ip4table`和`ip6table`，1001表示wan2从DHCP获取到的默认网关应该写入1001路由表。以及rule配置了路由规则，从lan2来的包都先查找1001路由表。如果wan2正常启动的话，那么1001路由表中就相应的默认网关，lan2来的包查找1001路由表的结果就是从wan2的网关发出去。

#### 双wan单lan，特定源IP路由到wan2范例：
```ini

config interface 'wan2'
        option proto 'dhcp'
        option device 'eth1'
        option ip4table '1001'
        option ip6table '1001'

config rule
        option in 'lan'
        option src '192.168.100.50/32'
        option lookup '1001'

```
其中的重点是wan2的`ip4table`和`ip6table`，1001表示wan2从DHCP获取到的默认网关应该写入1001路由表。以及rule配置了路由规则，从lan来的且源IP是'192.168.100.50/32'的包都先查找1001路由表。如果wan2正常启动的话，那么1001路由表中就相应的默认网关，查找1001路由表的结果就是从wan2的网关发出去。

Vash90, 2024-08-29:
感谢，大佬的回复。之前一直没搞定，都放弃了。
学习学习

closed 2024-08-29 by Vash90