istoreos/istoreos #1610: 登录系统后台时出现报错

wqinf, 2024-05-29:
/usr/lib/lua/luci/dispatcher.lua:784: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/dispatcher.lua:784: in function 'resolve_firstchild'
	/usr/lib/lua/luci/dispatcher.lua:864: in function 'resolve_page'
	/usr/lib/lua/luci/dispatcher.lua:888: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>

jjm2473, 2024-05-31:
是不是安装什么插件之后出现的？

wqinf, 2024-06-03:
> 是不是安装什么插件之后出现的？

这个就很难说了，因为插件安装的时候是正常的，并没有出现任何报错。

jjm2473, 2024-06-04:
就是安装插件以后，页面刷新就挂了吗

wqinf, 2024-06-04:
> 就是安装插件以后，页面刷新就挂了吗

并不是，是安装插件之后当时没有问题，第二天通过虚拟组网方式打开，发现突然报错了，但是不知道是哪一个插件导致的（也不确定是不是插件导致的），强制重启才好了。

jjm2473, 2024-06-04:
你的系统是装在u盘或者机械硬盘上吗？是不是掉盘了导致系统异常

wqinf, 2024-06-04:
> 你的系统是装在u盘或者机械硬盘上吗？是不是掉盘了导致系统异常

安装在固态硬盘（500G）中，但是安装了网心云插件（容器魔方），磁盘容量当时相对比较满（大约还剩30G的可用空间），不知道与这个是否有关，现在我给网心云插件（容器魔方）单独分了一个区（350G），再观察一下。

jjm2473, 2024-06-04:
下次出问题的话，可以打开ttyd（7681端口），然后执行下`dmesg | tail`。如果连ttyd都打不开，那大概率是系统盘出故障了。

wqinf, 2024-06-05:
> 下次出问题的话，可以打开ttyd（7681端口），然后执行下`dmesg | tail`看看内核有没有报错。如果连ttyd都打不开，那大概率是系统盘出故障了。

好的，不过这个ttyd我不太会用，是使用SSH工具直接输入软路由的ip，然后端口7681这样访问，然后执行`dmesg | tail`吗？能否指点一二，谢谢

liulu5468, 2024-06-05:
我也遇到了。前几天升级到了最近系统，没有安装任何新插件。
正常运行了2天了，当时一切正常。
现在打开samba一片空白，看不到任何东西，登录后台，发现错误页面。人在外面，还没办法重启。