istoreos/istoreos #1612: 单线复用时，首页无法显示接口状态

NaccOll, 2024-05-30:
## 1.关于你要提交的问题
在网口通过vlan单线复用时，首页不显示该网口，但OpenWrt自带的概览可以看到

## 2. 详细叙述

eth1虚拟eth1.2与eth1.3进行单线复用，eth1.2作为wan口，eth1.3作为lan口。上网情况正常，但首页不显示

### (2) 路由器型号和固件版本

BestCom N100 双网口小主机

### (3) 详细日志和/或截图

IStoreOS首页:
[image]

OpenWrt概览:
[image]

OpenWrt接口:
[image]

jjm2473, 2024-05-31:
那个只是vlan啊，vlan又没有物理参数，哪需要显示在网口状态里

jjm2473, 2024-05-31:
哦，是eth1没显示

NaccOll, 2024-06-01:
@jjm2473 是的，而且网口配置那边也显示连接上了，虽然判断不了网口对应的wan/lan状态，但是否可以将网口配置这边的展示搬运到首页

[image]

lucienyida, 2024-08-21:
你这个单线复用是单网口做拨号路由使用吗？   一直在找这个方法，