istoreos/istoreos #1613: 用Lucky做反代，在局域网内没法用域名访问内网端口，如何解决

HCNY2022, 2024-05-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
在istoreos的软路由上用的lucky做的https反代，443端口是开放，没有被运营商封禁，也没有使用端口转发（NAT回环在端口转发设置里），如图
[image]

外网用域名访问内网正常，内网也是用域名，但是连接不上
[image]

</body>
</html>
### (1) 具体问题

A：如何解决反代后，没有使用端口转发，如何实现局域网内域名访问内网服务
B:  nat回环应该只在设置端口转发的情况下才能去设置吧

### (2) 路由器型号和固件版本

A：
iStoreOS 22.03.6 2024052410 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa
### (3) 详细日志和/或截图

A：

138161, 2024-05-30:
dhcp/dns 主机名映射，不知道可不可以

HCNY2022, 2024-05-30:
> dhcp/dns 主机名映射，不知道可不可以

试过了，主机名映射不能设置ip+端口，只能域名对应ip，host文件也是这样，不能用。。。

138161, 2024-05-30:
先用http
再用ipv6
最后https+ipv6
瞎猜

jjm2473, 2024-05-31:
如果端口不一致，可以防火墙配置端口转发