istoreos/istoreos #1614: 三月系统莫名崩溃

next-step-jason, 2024-05-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86 64 n 5000

### (3) 详细日志和/或截图

A：用到1-2天的时候从文件共享模块开始崩溃，直到重启后也无法登录后台。

jjm2473, 2024-05-31:
可能中毒了吧

next-step-jason, 2024-06-04:
应该不是中毒，我换其他固件就没有问题，只有istore os有这个问题，就很迷惑。

next-step-jason, 2024-06-04:
Tue Jun  4 20:17:13 2024 daemon.err smbd[23250]: [2024/06/04 20:17:13.635379,  0] ../../lib/util/fault.c:305(log_stack_trace)
Tue Jun  4 20:17:13 2024 daemon.err smbd[23250]:   unable to produce a stack trace on this platform
Tue Jun  4 20:17:13 2024 daemon.err smbd[23250]: [2024/06/04 20:17:13.635450,  0] ../../source3/lib/dumpcore.c:283(dump_core)
Tue Jun  4 20:17:13 2024 daemon.err smbd[23250]:   Exiting on internal error (core file administratively disabled)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694291,  0] ../../source3/lib/util.c:490(reinit_after_fork)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   messaging_reinit() failed: NT_STATUS_DISK_FULL
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694411,  0] ../../source3/smbd/server.c:1024(smbd_accept_connection)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   reinit_after_fork() failed
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694472,  0] ../../lib/util/fault.c:173(smb_panic_log)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   ===============================================================
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694532,  0] ../../lib/util/fault.c:174(smb_panic_log)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   INTERNAL ERROR: reinit_after_fork() failed in pid 23251 (4.18.8)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694583,  0] ../../lib/util/fault.c:178(smb_panic_log)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   If you are running a recent Samba version, and if you think this problem is not yet fixed in the latest versions, please consider reporting this bug, see https://wiki.samba.org/index.php/Bug_Reporting
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694641,  0] ../../lib/util/fault.c:183(smb_panic_log)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   ===============================================================
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694689,  0] ../../lib/util/fault.c:184(smb_panic_log)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   PANIC (pid 23251): reinit_after_fork() failed in 4.18.8
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694743,  0] ../../lib/util/fault.c:305(log_stack_trace)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   unable to produce a stack trace on this platform
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]: [2024/06/04 20:17:14.694813,  0] ../../source3/lib/dumpcore.c:283(dump_core)
Tue Jun  4 20:17:14 2024 daemon.err smbd[23251]:   Exiting on internal error (core file administratively disabled)
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]: [2024/06/04 20:17:15.804752,  0] ../../source3/lib/util.c:490(reinit_after_fork)
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]:   messaging_reinit() failed: NT_STATUS_DISK_FULL
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]: [2024/06/04 20:17:15.804870,  0] ../../source3/smbd/server.c:1024(smbd_accept_connection)
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]:   reinit_after_fork() failed
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]: [2024/06/04 20:17:15.804929,  0] ../../lib/util/fault.c:173(smb_panic_log)
Tue Jun  4 20:17:15 2024 daemon.err smbd[23269]:    附上最近的日志，我只能看出来是Samba出了问题

jjm2473, 2024-06-04:
> failed: NT_STATUS_DISK_FULL

磁盘满了呗。是不是把文件下载到系统盘了

closed 2024-06-04 by jjm2473

next-step-jason, 2024-06-05:
文件下载外置硬盘的，我迁移一下docker试试吧

jjm2473, 2024-06-06:
那就看看哪个分区满了啊。要么就是系统盘满了，要么就是samba共享的文件夹所在分区满了。

next-step-jason, 2024-06-06:
系统分区和下载分区都没满，显示临时空间已满，请问怎么处理
[image]

jjm2473, 2024-06-06:
终端执行`cd /tmp; du -hd1`看看

next-step-jason, 2024-06-08:
看了下是用来上网的软件，但是刚需，不能删。请问有没有什么方法吗，定期清理tmp文件夹？

jjm2473, 2024-06-17:
这还能有什么办法，要么修复软件bug，要么看看这个软件能不能设置临时文件大小，要么就自己清理软件产生的临时文件，要么就看看能不能把这个软件的临时文件放在外部硬盘。