istoreos/istoreos #1619: 新版本DNS错误

QuantumLeap-us, 2024-05-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
升级新版本以后隔三岔五 DNS错误 上级路由为小米AX3000T 作为旁路由使用。怀疑是和passwall的dnsmasq冲突 请问如何解决。

### (2) 路由器型号和固件版本

A：设备型号
J4125   x86 Generic
固件版本
iStoreOS 22.03.6 2024052410
内核版本
5.10.201

### (3) 详细日志和/或截图

A：
[image]

closed 2024-06-05 by QuantumLeap-us

virtually321, 2024-08-31:
试试这两行命令解决dns问题
rm /etc/resolv.conf
ln -s /tmp/resolv.conf.auto /etc/resolv.conf

imrayyu, 2024-10-13:
同样的问题，请问怎么解决的呀？