istoreos/istoreos #1628: istoreos 里的istore 应用都无法安装

yuuionline, 2024-06-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：所有软件源都试过一遍，istore 应用都无法安装

### (3) 详细日志和/或截图

出现这个错误：Fetch feed list for aarch64_cortex-a53
curl: (6) Could not resolve host: istore.linkease.com

138161, 2024-06-10:
运营商dns服务器未覆盖linkease。com
安装smartdns
修改本机中：   /var/etc/smartdns/smartdns.conf
server-name smartdns
serve-expired yes
dnsmasq-lease-file /tmp/dhcp.leases
rr-ttl-min 600
log-size 64K
log-num 1
log-level error
resolv-file /tmp/resolv.conf.d/resolv.conf.auto
bind [::]:55333
bind-tcp [::]:55333
server-tls 223.5.5.5:853 
server-tls 120.53.53.53:853 
server-tls [2001:4860:4860::8888]:853 
server-tls 8.8.8.8:853 
server-https 223.5.5.5/dns-query 
server-https 120.53.53.53/dns-query 
conf-file /etc/smartdns/address.conf
conf-file /etc/smartdns/blacklist-ip.conf
conf-file /etc/smartdns/custom.conf

138161, 2024-06-10:
55333端口号可以自己修改

138161, 2024-06-10:
原理就是使用腾讯、阿里的加密dns，覆盖很全

jjm2473, 2024-06-17:
dns问题，istore.linkease.com是国内的域名，你这都解析不了，我也不知道说啥

closed 2024-06-17 by jjm2473

benjieming-0000, 2025-08-25:
wc, 一年了都没有修复

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/2200/1

benjieming-0000, 2025-10-26:
> [@benjieming-0000](https://github.com/benjieming-0000) 你确定现在有问题？你能打开这个网站吗 https://istore.linkease.com/repo/

我现在重新使用istore 后没有遇到这个问题了