istoreos/istoreos #1629: x86主机sys风扇频繁启停

Roiung, 2024-06-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
最近换了新的x86 nas主机，新主机带有cpu和sys两个温控风扇，bios中也有两个风扇分别的设置，均设置为低于16℃停止风扇，高于32℃启动风扇，60℃满速。安装windows或者在winpe中正常，sys风扇保持长时间正常转动状态，但在istoreos（已刷最新iStoreOS 22.03.6 2024052410）中，sys风扇频繁启停。因为web界面中看不到主板温度，从cpu温度处于52左右判断为主板温度低于50就停止，高于50启动。

### (1) 具体问题

A：如何在istoresos中能够使用bios中sys fan设置逻辑，保持32℃启动风扇，16℃停止。

### (2) 路由器型号和固件版本

A：x86主机，固件版本iStoreOS 22.03.6 2024052410

### (3) 详细日志和/或截图

A：

jjm2473, 2024-06-17:
x86并没有在系统里面控制风扇，我都不知道怎么控制。非x86的倒是可以在系统里控制。

Roiung, 2024-06-19:
我这几天仔细观察了下，发现SYS风扇并不是没有控制，虽然它在CPU温度50度左右会频繁启停，但是在CPU繁忙，CPU温度上去（比如60度以上）之后，SYS风扇会一直运行，而且转速会时高时低，噪音也就时大时小，一阵一阵的。而且从主页上看，CPU温度会跟着风扇噪音飘忽不定，瞬时最高温度会上到100多度，大部分时间目前室温20来度样子的话，CPU满负荷的时候CPU温度能在70度左右飘。

b0070, 2025-03-14:
联系增加风扇控制设置，发热严重，系统自己的风扇逻辑搞不懂。