istoreos/istoreos #1631: r2S - iStoreOS 22.03.6 2024052410 应用商店安装Jellyfin后，jallyfin路径的ffepmg进程占用异常，CPU 基本被跑满，卸载后正常，反复尝试问题依旧

calonye, 2024-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：应用商店安装Jellyfin后，jallyfin路径的ffepmg进程占用异常，CPU 基本被跑满，卸载后正常，反复尝试问题依旧

### (2) 路由器型号和固件版本

A：r2S - iStoreOS 22.03.6 2024052410 

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-06-17:
r2s还跑jellyfin？内存才多少

calonye, 2024-06-17:
> r2s还跑jellyfin？内存才多少

…你这关任务关的莫名其妙，r2s怎么不能跑了，我现在就跑起来了，电视装了kodi加j插件跑着4k还是能顺畅播放的，就你这种方式，还想着积极讨论呢，是不是有些主观的太想当然了？
经验主义不可取，经验只是辅助参考判断，不要直接一刀决断，这显得你很傲。

jjm2473, 2024-06-18:
所以呢，你希望开发者帮你做啥？

calonye, 2024-06-18:
> 所以呢，你希望开发者帮你做啥？

所以呢，你看到我在表达什么了吗？要求？期望？吐槽？埋怨？给你权限不是给你滥用

jjm2473, 2024-06-24:
那你就用r2s跑jellyfin呗，没人拦你啊。r2s就这点内存，而且还没硬解，体验好才怪了。

calonye, 2024-06-24:
> 那你就用r2s跑jellyfin呗，没人拦你啊。r2s就这点内存，而且还没硬解，体验好才怪了。

那你讲个屁，这么双标

jjm2473, 2024-06-26:
那我给你重新打开issue，满足你，看看有没有能帮你解答。

calonye, 2024-06-26:
[image]
目前闲置状态占用正常，播放时占用会较高，但满足基本使用；

原异常具体原因未具体探索分析，多次卸载后想起了docker镜像源的事，更换了另外一个源，并删除原遗留文件，重新下载安装，顺路安装完毕，经过设置刮削后，可正常播放“4k.PROPER.2160p.BluRay.REMUX.HEVC”资源；

也不知道这位网友急什么，这么着急下定义，你若是好心好意，我们就先就问题症状分析一波，你若是没那么闲，也请别这么武断就下定义。

原文“应用商店安装Jellyfin后，jallyfin路径的ffepmg进程占用异常，CPU 基本被跑满，卸载后正常，反复尝试问题依旧”

我说的算比较明白了，也没有哪句让你觉得表达成了“伸手党”。没搞懂你的脑回路，或许你只是习惯性跳着字眼看，算了，不想纠结，祝你生活愉快。

CoreJK, 2025-02-21:
#

> [image] 目前闲置状态占用正常，播放时占用会较高，但满足基本使用；
>
> 原异常具体原因未具体探索分析，多次卸载后想起了docker镜像源的事，更换了另外一个源，并删除原遗留文件，重新下载安装，顺路安装完毕，经过设置刮削后，可正常播放“4k.PROPER.2160p.BluRay.REMUX.HEVC”资源； [...]

请教一下，你用的是那个 docker 镜像，我也是 r2s ，在尝试安装 jellfin 的 dcoker 版本
尝试了 rockchip 版本，也是 cpu 占用很高😥