istoreos/istoreos #1632: rtw_8821cu无线网卡模块运行异常，大佬帮忙看看

xiaomeng9597, 2024-06-12:
错误日志

`Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]: ttyd 1.6.3 (libwebsockets 4.2.1-unknown)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]: tty configuration:
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:   start command: /bin/login
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:   close signal: SIGHUP (1)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:   terminal type: xterm-256color
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]: LWS: 4.2.1-unknown, loglevel 7
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]: NET CLI SRV H1 H2 WS ConMon IPV6-off
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:    /usr/lib/libwebsockets-evlib_uv.so
Tue Jun 11 17:36:27 2024 daemon.err ttyd[11686]: [2024/06/11 17:36:27:8746] N:  Using foreign event loop...
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:  ++ [wsi|0|pipe] (1)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:  ++ [vh|0|netlink] (1)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:  ++ [vh|1|default||7681] (2)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]: lws_socket_bind: nowsi: source ads 0.0.0.0
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:  ++ [wsi|1|listen|default||7681] (2)
Tue Jun 11 17:36:27 2024 daemon.notice ttyd[11686]:  Listening on port: 7681
Tue Jun 11 17:36:28 2024 daemon.err nmbd[9415]: [2024/06/11 17:36:28.535099,  0] ../../source3/nmbd/nmbd.c:901(main)
Tue Jun 11 17:36:28 2024 daemon.err nmbd[9415]:   nmbd version 4.18.8 started.
Tue Jun 11 17:36:28 2024 daemon.err nmbd[9415]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Tue Jun 11 17:36:28 2024 daemon.info [7384]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe300000.ethernet': not supported by any plugin
Tue Jun 11 17:36:29 2024 daemon.info [7384]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe380000.usb/usb1/1-1/1-1.1': not supported by any plugin
Tue Jun 11 17:36:30 2024 daemon.err smbd[9414]: [2024/06/11 17:36:30.439950,  0] ../../source3/smbd/server.c:1746(main)
Tue Jun 11 17:36:30 2024 daemon.err smbd[9414]:   smbd version 4.18.8 started.
Tue Jun 11 17:36:30 2024 daemon.err smbd[9414]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Tue Jun 11 17:36:32 2024 daemon.info procd: - init complete -
Tue Jun 11 17:36:41 2024 daemon.err dockerd[10345]: time="2024-06-11T09:36:41.836465007Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Tue Jun 11 17:36:42 2024 daemon.err dockerd[10345]: time="2024-06-11T09:36:42Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Tue Jun 11 17:36:42 2024 daemon.err modprobe: failed to find a module named aufs
Tue Jun 11 17:36:42 2024 daemon.err dockerd[10345]: time="2024-06-11T09:36:42.563004824Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Tue Jun 11 17:36:42 2024 daemon.err dockerd[10345]: time="2024-06-11T09:36:42.564410944Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Tue Jun 11 17:36:42 2024 daemon.err dockerd[10345]: time="2024-06-11T09:36:42.572033587Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Tue Jun 11 17:36:42 2024 kern.warn kernel: [   48.915391] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:42 2024 kern.warn kernel: [   48.915405] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:42 2024 kern.warn kernel: [   48.915414] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:42 2024 kern.warn kernel: [   48.915421] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:42 2024 daemon.info modprobe: nf_conntrack is already loaded
Tue Jun 11 17:36:42 2024 daemon.info modprobe: nf_conntrack_netlink is already loaded
Tue Jun 11 17:36:42 2024 daemon.info avahi-daemon[7319]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Tue Jun 11 17:36:42 2024 daemon.info avahi-daemon[7319]: New relevant interface docker0.IPv4 for mDNS.
Tue Jun 11 17:36:42 2024 daemon.info avahi-daemon[7319]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Tue Jun 11 17:36:43 2024 daemon.debug ModemManager[12380]: hotplug: event reported: action=add, name=docker0, subsystem=net
Tue Jun 11 17:36:43 2024 daemon.err ModemManager[12380]: hotplug: parent device sysfspath not found
Tue Jun 11 17:36:43 2024 kern.warn kernel: [   49.612953] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:43 2024 kern.warn kernel: [   49.612965] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:43 2024 kern.warn kernel: [   49.612972] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:43 2024 kern.warn kernel: [   49.612977] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
Tue Jun 11 17:36:46 2024 daemon.err uwsgi[7424]: *** has_emperor mode detected (fd: 7) ***
Tue Jun 11 17:36:46 2024 daemon.err uwsgi[7424]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
Tue Jun 11 17:36:46 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:36:46 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Tue Jun 11 17:36:46 2024 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 12450)
Tue Jun 11 17:36:46 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:36:46 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Tue Jun 11 17:36:46 2024 daemon.err nginx[7490]: 2024/06/11 17:36:46 [error] 7517#0: *3 open() "/www/favicon.ico" failed (2: No such file or directory), client: 192.168.23.238, server: _lan80, request: "GET /favicon.ico HTTP/1.1", host: "192.168.23.112", referrer: "http://192.168.23.112/"
Tue Jun 11 17:36:47 2024 daemon.err nginx[7490]: 2024/06/11 17:36:47 [error] 7517#0: *3 open() "/www/favicon.ico" failed (2: No such file or directory), client: 192.168.23.238, server: _lan80, request: "GET /favicon.ico HTTP/1.1", host: "192.168.23.112", referrer: "http://192.168.23.112/"
Tue Jun 11 17:36:47 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:36:47 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Tue Jun 11 17:36:54 2024 user.info : luci: accepted login on / for root from 192.168.23.238
Tue Jun 11 17:36:58 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:36:58 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Tue Jun 11 17:37:12 2024 daemon.info uwsgi-luci: Tue Jun 11 17:37:12 2024 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/istore/u/guide/ddns/ (ip 192.168.23.238) !!!
Tue Jun 11 17:37:12 2024 daemon.info uwsgi-luci: Tue Jun 11 17:37:12 2024 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/istore/u/guide/ddns/ (192.168.23.238)
Tue Jun 11 17:37:16 2024 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Tue Jun 11 17:37:56 2024 daemon.info hostapd: wlan0: STA 46:7c:6b:cc:de:b0 IEEE 802.11: authenticated
Tue Jun 11 17:37:56 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 46:7c:6b:cc:de:b0 1
Tue Jun 11 17:37:56 2024 daemon.info hostapd: wlan0: STA 46:7c:6b:cc:de:b0 IEEE 802.11: associated (aid 1)
Tue Jun 11 17:37:56 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 46:7c:6b:cc:de:b0
[... cut ...]
Tue Jun 11 17:55:54 2024 daemon.err uwsgi[7424]: *** has_emperor mode detected (fd: 8) ***
Tue Jun 11 17:55:54 2024 daemon.err uwsgi[7424]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-cgi_io.ini
Tue Jun 11 17:55:54 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:55:54 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini has been spawned
Tue Jun 11 17:55:54 2024 daemon.err uwsgi[7424]: Tue Jun 11 17:55:54 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is ready to accept requests`

xiaomeng9597, 2024-06-12:
系统刚重启后wlan0正常，有WiFi信号出来，用设备链接WiFi一会后就掉线了，然后报上面的错误，不知道咋解决了

closed 2024-06-20 by xiaomeng9597

peanutlya, 2024-08-13:
同样的问题，后面解决了吗

peanutlya, 2024-08-13:
> 系统刚重启后wlan0正常，有WiFi信号出来，用设备链接WiFi一会后就掉线了，然后报上面的错误，不知道咋解决了

同样的问题，后面解决了吗