istoreos/istoreos #1635: 磁盘扩容后，istore对于新挂载硬盘剩余容量不显示

NaccOll, 2024-06-12:
## 1.关于你要提交的问题

磁盘扩容后，istore对于新挂载硬盘剩余容量不显示

## 2. 详细叙述

pve新增10g磁盘给openwrt，进行磁盘扩容，采用更换/挂载点的方式扩容，目前openwrt的概览显示正常，istorex首页无法显示剩余容量

### (2) 路由器型号和固件版本

BestCom N100 双网口小主机

### (3) 详细日志和/或截图

[image]

[image]

[image]

jjm2473, 2024-06-17:
/是没法直接更换的，除非你把整个根文件系统都拷过去。

jjm2473, 2024-06-17:
如果你确实知道extroot，那么参考这个教程 https://blog.chaos.run/dreams/openwrt-extroot/index.html

如果你连extroot都不知道干嘛的，那就别试了。我怀疑你根本就试瞎搞。如果你只是想换个大点的磁盘，那应该使用克隆的方式，或者执行quickstart命令将系统安装到新磁盘上。

closed 2024-06-17 by jjm2473

NaccOll, 2024-06-17:
@jjm2473 我这个issue想问的就是这个，为什么我这么操作后，我的容量扩大了，软件包也扩大了，但istore主页不显示并且仍然提示我需要格式化呢？因为我是openwrt原版装的istore的软件中心，不知道是否与这个有关

我参考的是block-mount的教程，如下，只是把路径替换了
[image]

**修改的效果如下，的确扩大了**
[image]
[image]
[image]

但现在就是istorex的主页不显示
[image]

jjm2473, 2024-06-18:
可能因为你没分区吧，你是直接把整个磁盘格式化了，quickstart识别不了这种不正常的磁盘。

jjm2473, 2024-06-18:
还有，你这样搞的话，之后就不用升级固件了，反正升级了也没用。