istoreos/istoreos #1640: i226 v网卡的4口软路由，网络会从1000M变成100M 光猫是千兆接口

xinzhongyouhai, 2024-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

## 2. 详细叙述

### (1) 具体问题
i226 v网卡的4口软路由，网络会从1000M变成100M 光猫是千兆接口

### (2) 路由器型号和固件版本
<html>
<body>

主机名 | iStoreOS
-- | --
型号 | Default string Default string
架构 | Intel(R) N100
目标平台 | x86/64
固件版本 | iStoreOS 22.03.6 2024052410 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

</body>
</html>

### (3) 详细日志和/或截图
内核日志
[673462.275576] igc 0000:01:00.0 eth0: NIC Link is Down
[673470.555600] igc 0000:01:00.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
[673480.865552] pppoe-wan: renamed from ppp0
[673488.942397] passwall: restart when wan ifup
[673496.985974] passwall: restart when wan_6 ifup
[673504.002772] passwall: restart when wan_6 ifup
[673705.226187] igc 0000:01:00.0 eth0: NIC Link is Down
[673716.066096] igc 0000:01:00.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[673726.916145] pppoe-wan: renamed from ppp0
[673733.948422] passwall: restart when wan ifup
[673741.964388] passwall: restart when wan_6 ifup
A：

到底是i226自身问题还是驱动问题？如果是驱动问题可以更新驱动吗？

xinzhongyouhai, 2024-06-14:
后来我重新插拔了WAN口网线 就会恢复到1000M连接~不是第一次遇到这种情况了~

hza1128, 2024-06-16:
兼容问题，用此法解决：https://www.right.com.cn/forum/thread-8284175-1-1.html

xinzhongyouhai, 2024-06-23:
> 兼容问题，用此法解决：https://www.right.com.cn/forum/thread-8284175-1-1.html

你给的这个方案有个问题~路由重启后 网口无法连接

dunkeng, 2024-06-24:
换个网线就好了

xinzhongyouhai, 2024-06-24:
> 换个网线就好了

根网线没关系~实际上插拔下网线有几率恢复~但是这是治标不治本的方案~

JiangJamm, 2024-06-24:
同J4125+4网口i226-v，也遇到断流问题，降温也没办法解决。
有看到让关闭TSO的：https://www.chiphell.com/thread-2535386-1-1.html
还有说内核6.1的没有问题的。
暂时还没尝试能否解决

xinzhongyouhai, 2024-06-24:
> 同J4125+4网口i226-v，也遇到断流问题，降温也没办法解决。 有看到让关闭TSO的：https://www.chiphell.com/thread-2535386-1-1.html 还有说内核6.1的没有问题的。 暂时还没尝试能否解决

这就是传说中的断流？？？

jjm2473, 2024-07-08:
换网线

JiangJamm, 2024-07-13:
> 换网线

太抽象了，我一直对我的网线很有信心，因为是新买的小米路由器送的超五类，没想到自己换了一条网线，真的稳定了😂

xinzhongyouhai, 2024-07-15:
> > 换网线
> 
> 太抽象了，我一直对我的网线很有信心，因为是新买的小米路由器送的超五类，没想到自己换了一条网线，真的稳定了😂

情况会复现的~我试过~根网线没关系~

JiangJamm, 2024-07-15:
> > > 换网线
> > 
> > 
> > 太抽象了，我一直对我的网线很有信心，因为是新买的小米路由器送的超五类，没想到自己换了一条网线，真的稳定了😂
> 
> 情况会复现的~我试过~根网线没关系~

[image]

我的挺稳定的，更换了网线之后已经稳定运行了4天没有宕机了

jjm2473, 2024-07-17:
可能这个驱动比较挑网线。如果不确定是否网线问题，就找根千兆扁网线试试，扁线一般质量比较好。

hunjixin, 2025-02-22:
也有类似的问题，重启之后有时候显示千兆，过一会自动变成百兆

JiangJamm, 2025-02-22:
> > > > 换网线
> > > 
> > > 
> > > 太抽象了，我一直对我的网线很有信心，因为是新买的小米路由器送的超五类，没想到自己换了一条网线，真的稳定了😂
> > 
> > 
> > 情况会复现的~我试过~根网线没关系~
> 
> <img alt="image" width="647" src="https://private-user-images.githubusercontent.com/39955422/348774494-e0bcdd5c-45d2-461e-8705-f3e0435f5877.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAyMjUyNTYsIm5iZiI6MTc0MDIyNDk1NiwicGF0aCI6Ii8zOTk1NTQyMi8zNDg3NzQ0OTQtZTBiY2RkNWMtNDVkMi00NjFlLTg3MDUtZjNlMDQzNWY1ODc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjIyVDExNDkxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhMTJhMDA5YWNmZTFhODZlYWY1ODYxYzY1NmJhNjUyNzNlNTQwODNlODNiOGQwNjBlNjQzYTdjZWEzMDBkYjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Ly02TrI0akr4sG6oqowFtvjfHPowbhhk0vfBVBIJS_U">
> 我的挺稳定的，更换了网线之后已经稳定运行了4天没有宕机了

最近我又开始出现断网和降速问题了，我都怀疑我的硬件是不是烧网线了