istoreos/istoreos #1643: 每次重启后无法自动获取ipv6地址，需要到接口手动重启wan口

Theo-messi, 2024-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每次重启后无法自动获取ipv6地址，需要到接口手动重启wan口

### (2) 路由器型号和固件版本

A：iStoreOS 22.03.6, 2024052410

### (3) 详细日志和/或截图

这是路由器重启后的

[image]

手动重启wan口后就显示正常

[image]

jjm2473, 2024-06-17:
可能运营商限制了重拨ipv6的间隔

Theo-messi, 2024-06-17:
> 可能运营商限制了重拨ipv6的间隔

应该是我之前的设置有问题，我重置了系统又重新设置了下，现在恢复正常了

closed 2024-06-17 by Theo-messi