istoreos/istoreos #1644: 系统似乎没有k10temp?

kuoluo1995, 2024-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我发现我的电脑安装了istoreos后没办法显示温度

### (2) 路由器型号和固件版本

A：路由器型号：Thinkbook 14 2020 (amd r7 5800U)
B：固件：iStoreOS 22.03.6 2024061415 （x86_64_efi）

### (3) 详细日志和/或截图

root@iStoreOS:/# sudo opkg list | grep kmod-hwmon
kmod-hwmon-core - 5.10.201-1
kmod-hwmon-drivetemp - 5.10.201-1
kmod-hwmon-it87 - 5.10.201-1
kmod-hwmon-pwmfan - 5.10.201-1
kmod-hwmon-vid - 5.10.201-1
root@iStoreOS:/# sudo opkg list | grep k10temp
root@iStoreOS:/# sudo modprobe drivetemp
d
sensors
sudo modprobe it87
sudo modprobe pwmfan
sudo modprobe vid
sensors
root@iStoreOS:/# sensors
amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:      806.00 mV
vddnb:       824.00 mV
edge:         +51.0°C
power1:        2.00 W

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A

我sensors没有显示温度，看之前的issue说似乎需要有k10temp。但我发现我的系统里似乎也没有k10temp？

jjm2473, 2024-06-17:
可以自己安装驱动，反正x86是兼容openwrt的内核的，如果openwrt那边有驱动，你就可以直接在软件包里面安装

jjm2473, 2024-06-17:
题外话，openwrt里面不需要sudo，默认都是root用户执行的

closed 2024-06-17 by jjm2473

kuoluo1995, 2024-06-17:
你好，我尝试去下载，但发现并没有找到对应的驱动。以及这似乎是linux内核里面的？
我看历史的问题中，别人似乎本来就包含了这个k10temp驱动。
https://github.com/istoreos/istoreos/issues/982?ref=https://codemonkey.link

现在最新一版里是不是去掉了？

kuoluo1995, 2024-06-17:
> 可以自己安装驱动，反正x86是兼容openwrt的内核的，如果openwrt那边有驱动，你就可以直接在软件包里面安装
你好，我尝试去下载，但发现并没有找到对应的驱动。以及这似乎是linux内核里面的？
我看历史的问题中，别人似乎本来就包含了这个k10temp驱动。
https://github.com/istoreos/istoreos/issues/982?ref=https://codemonkey.link

现在最新一版里是不是去掉了？

jjm2473, 2024-06-18:
那就先试试openwrt官方固件，有问题先反馈给他们，x86的这些基础驱动都是openwrt官方的，istoreos修改或者增加的都是些网络/GPU驱动