istoreos/istoreos #1645: PVE 内istoreos重启设备之后，插件丢失

yjc980121, 2024-06-17:
反馈bug/问题模板，提建议请删除

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
pve内安装istoreos，安装了很多插件，重启设备之后插件全部丢失
开始pve设置的单网口eth0，后来增加一个网口eth1 ,看描述单网口是作为lan口，双网口，eth0是wan口，其他是lan口，虚拟机没有其他改动。
A：

### (2) 路由器型号和固件版本
istoreos-21.02.1-2022042317-x86-64-generic-image.iso
A：

### (3) 详细日志和/或截图

A：

rayangl, 2024-06-17:
我是重启以后沙箱丢失，磁盘管理里面有，首页的沙箱按钮变成了开启沙箱，还要格式化，我想重新挂载，怕整错了 数据没了

yjc980121, 2024-06-18:
知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了

closed 2024-06-18 by yjc980121

hadis898, 2024-09-03:
> 知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了

兄弟你怎么解决的，我也遇到这种问题了

yjc980121, 2024-09-18:
> > 知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了
> 
> 兄弟你怎么解决的，我也遇到这种问题了

把CD光驱去掉呀，引导从光驱改为硬盘就行了，或者调整引导顺序，硬盘放前边

who10086, 2025-07-30:
> > > 知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了
> > 
> > 
> > 兄弟你怎么解决的，我也遇到这种问题了
> 
> 把CD光驱去掉呀，引导从光驱改为硬盘就行了，或者调整引导顺序，硬盘放前边

已经去掉了，每次重启都丢失所有东西

yjc980121, 2025-08-01:
> > > > 知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了
> > > 
> > > 
> > > 兄弟你怎么解决的，我也遇到这种问题了
> > 
> > 
> > 把CD光驱去掉呀，引导从光驱改为硬盘就行了，或者调整引导顺序，硬盘放前边
> 
> 已经去掉了，每次重启都丢失所有东西

那就是你没安装到硬盘上，或者你克隆了一个livecd的，正确安装下

yjc980121, 2025-08-05:
> > > > > > 知道问题了，虚拟机启动引导还是iso镜像，更换硬盘后好了
> > > > > 
> > > > > 
> > > > > 兄弟你怎么解决的，我也遇到这种问题了
> > > > 
> > > > 
> > > > 把CD光驱去掉呀，引导从光驱改为硬盘就行了，或者调整引导顺序，硬盘放前边
> > > 
> > > 
> > > 已经去掉了，每次重启都丢失所有东西
> > 
> > 
> > 那就是你没安装到硬盘上，或者你克隆了一个livecd的，正确安装下
> 
> 通过quick里面的install x86选项也没装上吗？那个选项提示已经安装成功了然后让我删了镜像，还要怎么操作才算安装上呢？
那就已经安装上了，剩下的就是把光盘弹出去，如果是虚拟机，最好连驱动器都删了，有些虚拟机，引导顺序，是光盘在前