istoreos/istoreos #1647: R2S最新固件80和53端口被占用，无法使用AdguardHome

bzkmsjy, 2024-06-17:
版本StoreOS 22.03.6 2024061415
安装ADG启用后无法进入管理页面，有红字validating ports: listen tcp 0.0.0.0:80: bind: address already in use和validating ports: listen tcp 0.0.0.0:53: bind: address already in use

奇怪的是我另一个斐讯N1 240430版本的系统没有这个问题

两台设备的使用方法完全一致，只安装了网心云、AdguardHome和OpenClash三个插件，作为旁路由的IP和网关也一毛一样（当然没有同时开机），N1完美使用，R2S却出现了端口被占用。

虽然这个情况应该有解决办法，但是iStoreOS作为面向小白的系统，请不要对我的动手能力有什么期待，完全看不懂。。。

bzkmsjy, 2024-06-17:
这是个乌龙，根本不需要释放80和53端口，自己改别的就行，默认设置很蠢，明明luci设置里都把管理网页端口改成3000了，景区后还是默认80，ADG官方还在错误警告下面放了一个如何释放53端口的解决方法，更加误导人了

用了1年多OP，我才知道以前用的AdguardHome都是固件/软件包里就自带了发布者的自定义设置的，导致我其实从来没有从零开始配置过ADG……

然后R2S和N1的固件确实不是同一个地方下的……可能自带了一些区别。

有时候太照顾小白反而不好，让我们从头多尝试比较好。

jjm2473, 2024-06-18:
同一个作者的固件应该一样的

closed 2024-06-18 by jjm2473

zifeng-radxa, 2025-07-30:
> 这是个乌龙，根本不需要释放80和53端口，自己改别的就行，默认设置很蠢，明明luci设置里都把管理网页端口改成3000了，景区后还是默认80，ADG官方还在错误警告下面放了一个如何释放53端口的解决方法，更加误导人了
> 
> 用了1年多OP，我才知道以前用的AdguardHome都是固件/软件包里就自带了发布者的自定义设置的，导致我其实从来没有从零开始配置过ADG……
> 
> 然后R2S和N1的固件确实不是同一个地方下的……可能自带了一些区别。
> 
> 有时候太照顾小白反而不好，让我们从头多尝试比较好。

请问不释放 53 是随意改也可以吗，需要 DNS 转发吗